rstrouse/ESPSomfy-RTS #145: Unable to connect to device after upgrading to 2.1.4

FlaMike, 2023-09-03:
I've been running 2.1.3 since it was released 3 weeks ago, along with the companion version of the HA  integration.  This morning I noticed the HA  integration had an update, so I installed version 2.1.4 on the device, to be followed by updating the HA integration.  After following the usual device upgrade procedure (download the firmware & app to a PC, update the firmware, update the app--saving a backup copy of the settings, & waiting for the device to restart) I was not able to reconnect to the device; I just got the spinning circle of bubbles.  After a few minutes, I unplugged the device's power & the plugged it back in.  My router showed it was connected, but I could not access the UI.  Similarly, I could not access it via the HA integration.

I power cycled a couple of times with no love from the UI.  Eventually, I re-flashed 2.1.3 onto the device,  reconfigured the network settings & time zone, reloaded the backup, enabled the radio, & rebooted the device.  Whew!  Back to "normal" on 2.1.3.  

I have no idea why I wasn't able to access the device under 2.1.4.  I also don't know if others have had a similar issue.  For the moment, I'm fine on 2.1.3, but I wanted you to know there *may* be a problem with 2.1.4.  I'm fine on 2.1.3 and the new features in 2.1.4 are not needed for my happy & continued use of RTS Somfy.

rstrouse, 2023-09-03:
What do you mean by unable to connect?  Did your browser give you a not found error, display a blank page, or something else?

FlaMike, 2023-09-03:
The browser gave me a blank page & spinning cursor.  The page never loaded, but the router said ESP Somfy was connected to the network.  Device's AP went away, of course.

rstrouse, 2023-09-03:
I am pretty sure this has nothing to do with the code but more likely with the flashing of the file system using the ESP OTA functions.  It is almost as if the esp library did not commit the page files and configuration to the flash partition.  Can I get you to make a backup of your configuration and try again?

FlaMike, 2023-09-03:
I'll be happy to try again with a fresh download of the updates.  It may be later today or tomorrow before I can give it a shot.  Hope that's OK at your end!

FYI, when I reflashed 2.1.3, I downloaded the "onboarding" version of ESP-Somfy & flashed it via web.esphome.io as if it was a brand new installation.  Since I couldn't access the device, I thought that was  the most direct path to recovery.  I don't know if the onboarding binary for 2.1.4 would have gotten me back on track as well.

rstrouse, 2023-09-03:
Yep that will work.  I suspect that using the 2.1.4 onboard file would have worked as well.  If I can figure out what is missing perhaps I can force the OTA library to roll back.

wutz-ESP, 2023-09-04:
One device NodeMCU ESP-32S and two WT32-ETH01 updated from 2.1.3 without any problem.

FlaMike, 2023-09-04:
Tried upgrading this morning from 2.1.3 to 2.1.4 with fresh downloads of both the firmware & application.  Worked without a hitch, as well as the HA integration upgrade.  Don't know why it failed yesterday.  Bad download perhaps?  I don't know if there's a checksum on the downloads & I had a bit or three flipped along the way.

One thing that was a bit strange was the screen after upgrading the application (see attachment).  I refreshed  the screen & everything was normal.  Bad spirits in the area?

Anyway, all good here & functioning normally, so I'll close this issue.  Thanks for your quick responses yesterday.  Have a wonderful holiday!

[image]

closed 2023-09-04 by FlaMike