rstrouse/ESPSomfy-RTS #146: Shutter that does not want to program itself

elpigeondindo, 2023-09-07:
I installed this system on an esp32 with a CC1101. It worked really well for 4 of my shutters without any problems.
But for one, it is impossible to jog it when I press PROG. I tried to modify the frequencies but it doesn't change anything.
I then factory reset the shutter but that doesn't change anything.

[image]

What can I do to add it?

rstrouse, 2023-09-07:
First, if you are running an E07-M1101 it is only capable of 10dBm transmit power.  If you are running the most recent version press and hold the prog button in ESPSomfy RTS for up to 2 seconds or until it jogs.  Another user had a tardy shade.  Also, try moving the device closer to the motor to see if it pairs.

elpigeondindo, 2023-09-08:
I tried to hold down the prog button but it didn't change anything. I also brought the esp closer but the same nothing.
I think it must be at the level of the frequencies that I should modify but I don't know what to put

rstrouse, 2023-09-09:
Is there some difference between these motor models?

elpigeondindo, 2023-09-09:
Oui c'est exactement les meme models. C'est pour ca que je compreds pas pourquoi celui la ne veut pas se synchroniser.

Yes it's exactly the same models. That's why I don't understand why this one doesn't want to synchronize.

rstrouse, 2023-09-09:
That is very odd.  Does your remote list the frequency on the back?  If it does set the Base Frequency slider to that frequency listed on the remote.

elpigeondindo, 2023-09-10:
no there is no frequency on the back. However it works perfectly with my RFXtrx433XL. Besides, I have 2 terrace blinds that are not even seen by the software, while they also work very well with the RFX

rstrouse, 2023-09-10:
RTS PURE should be picked up readily by ESPSomfy RTS.  Do you see any responses in the radio log for these two terrace blinds?  Can you please select the channel that is controlling one of these blinds that cannot be controlled and copy the log here.

elpigeondindo, 2023-09-11:
Yes I can ^^ but exactly how can I do that?

rstrouse, 2023-09-11:
Click the copy button on this screen and paste it into a message here.  That will copy all the pulse timings from the frames in the logs into a message.

[image]

elpigeondindo, 2023-09-11:
ESPSomfy-RTS only sees the up, down or stop commands but the PROG command. Here is what I have in log when I wanted to make a prog.

[{"encKey": 172,
  "address": 14048562,
  "rcode": 2500,
  "command": "Down",
  "rssi": -75,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [2536,2505,2517,2502,4899,1256,1311,1248,1339,1269,672,609,1316,620,682,1246,693,630,661,616,1345,1224,684,626,1319,1278,667,599,682,637,671,593,1327,622,668,1264,1333,617,680,591,679,637,673,1246,1328,631,687,611,673,601,675,616,709,1226,703,379,505,77,771,243,266,606,673,583,1353,1286,668,619,1325,1267,1306,1243,1360,1251,1334,1245,700,608,662,616,1324,611,688,1267,671,611,1308,615,696,1268],
  "time": "2023-09-11T19:03:15.294+0200"
},
{"encKey": 160,
  "address": 14048562,
  "rcode": 2504,
  "command": "Down",
  "rssi": -78,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [66297,14396,10260,17753,2514,2507,2557,2489,4893,1269,1298,1271,1328,602,655,631,671,629,653,635,663,1276,679,632,639,612,1327,615,680,635,650,639,656,1280,677,616,676,621,671,631,1310,1263,1307,639,648,633,665,643,647,628,678,1265,1323,608,682,601,673,635,659,629,667,1275,688,590,674,607,705,615,1316,1255,684,628,1293,1291,1298,1289,1307,1265,1306,1286,663,618,672,633,1313,615,698,1252,681,75,125,607,1319,632,673,1263],
  "time": "2023-09-11T19:03:57.302+0200"
},
{"encKey": 161,
  "address": 14048562,
  "rcode": 2505,
  "command": "My",
  "rssi": -79,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [66061,122856,56327,8721,57847,122840,10504,485,8673,49596,10285,17769,2491,2539,2507,2526,4846,1270,1334,1266,1311,620,649,654,640,644,647,1279,667,646,1308,1283,652,643,1287,1274,1287,26,1315,630,629,1309,1293,653,620,669,638,683,619,1320,603,666,620,671,1281,666,629,664,635,1304,1282,1284,1300,633,653,1320,1274,1302,629,683,601,673,1254,680,608,699,605,707,558,688,625,671,613,686,632,647,25,618,678,630,1319,1273,657,640,1324,616,659,1274,649,639,1328,599,675],
  "time": "2023-09-11T19:03:58.061+0200"
},
{"encKey": 162,
  "address": 14048562,
  "rcode": 2506,
  "command": "Up",
  "rssi": -69,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [63278,5586,121907,119229,593,581,3517,2254,73,1913,101,7623,10294,17726,2540,2533,2511,2516,4849,1272,1309,1272,1327,616,669,633,657,1281,1319,1262,1321,610,685,621,667,609,689,1251,1322,1272,670,630,1311,619,668,611,684,1276,646,636,1310,618,676,637,657,1268,1330,615,671,628,659,1269,669,629,1311,1272,1306,624,670,1258,1334,622,679,608,681,592,699,1269,650,618,1330,616,672,1276,663,614,1320,1263,676,619,676,613,684,619,668,636,660,616,675,619,680,617,679,611],
  "time": "2023-09-11T19:04:01.744+0200"
}]

elpigeondindo, 2023-09-11:
yes the usual procedure. I have a telis remote

- I lower the shutter a little
- I press the back of the remote control until the shutter moves
- I press several times on prog of the ESPSomfy-RTS
- To finish the program I press the button again on the back of the remote control and the shutter starts to move again.

But with all that there is only the moment when I bring down the pane that appears in the logs. No info during programming. I have the exact same pane on another window and it worked straight away.

rstrouse, 2023-09-11:
That is odd.  You do not see the prog commands coming from the remote on that channel?  Which Telis do you have is it a 4 or a 16?

elpigeondindo, 2023-09-11:
I do not know at all. Behind it says TELIS 1 / RTS PURE

elpigeondindo, 2023-09-11:
So I stood right next to the esp. I tested the problematic pane and nothing appears.

I tried another remote control and here is the result:

[{"encKey": 163,
  "address": 10004534,
  "rcode": 803,
  "command": "Prog",
  "rssi": -25,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [16341,104476,85923,10253,18114,2493,2525,2504,2523,4818,1270,1281,1279,1294,641,664,620,644,1283,653,643,1275,640,667,1273,1280,641,646,1295,638,641,1291,648,634,638,640,1296,1281,640,640,25,1295,1288,1276,1301,640,640,644,647,634,659,640,641,1278,641,648,1295,1265,1290,649,633,1291,643,648,636,638,654,644,1272,641,665,619,669,1260,656,640,1300,1263,646,655,1291,619,670,633,635,640,647,644,654,639,635,643,642,632,660,642,653],
  "time": "2023-09-11T20:39:02.194+0200"
}]

elpigeondindo, 2023-09-11:
It's exactly the same remote control with the same references.

rstrouse, 2023-09-11:
Expand the RX Bandwidth 50kHz at a time until we hear the prog button.  It is so odd that the motor can hear that remote but it clearly isn't transmitting around the center band.   You said if you press and hold the prog button on that remote it causes the shutter to jog (just verifying).

elpigeondindo, 2023-09-11:
Yes with the remote control, programming takes place without any problem

elpigeondindo, 2023-09-11:
Great, we're moving forward. Now I see the remote control prog button appear in the logs. I tried to make the esp prog but it still didn't work. I am in 108Kz

rstrouse, 2023-09-11:
Just to be sure you have the TX Power set to 10dBm.  It did send a prog command but it is not being heard by the motor.

First lets verify that you are sending an RTS command.  Check to see that you have set the protocol to RTS on the shutter config and the bit length is 56-bits.

elpigeondindo, 2023-09-11:
So I checked yes TX on 10db, RTS and 56bits.

When I press the prog button on the ESP, should we see the trace in the logs? because nothing is displayed for me

rstrouse, 2023-09-11:
No you should not see transmitted frames in the log.  It will only reflect output from the receiver side of the transceiver.  If you had two ESPSomfy RTS devices one would show the receive side.  TX output is typically not an issue since the CC1101 can transmit with more power than a remote.

elpigeondindo, 2023-09-11:
so I tried the prog again.

I pressed 3 seconds on the remote control, then I pressed 30 times and stayed 2 seconds on the prog button of the esp but nothing

[{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 4,
  "pulses": [246067,245333,124315,121646,127085,120385,65428,10293,17716,2534,2545,2486,2525,4861,1261,1320,1267,1325,610,687,1247,668,634,1312,620,676,614,665,1279,1324,1257,1320,1258,1319,636,660,613,669,1286,1310,620,682,605,664,1268,684,614,666,632,659,632,679,617,653,637,656,618,669,638,1313,1271,1311,616,672,1258,1317,1273,1312,1271,681,611,1306,1281,682,621,668,606,1310,1268,683,616,1306,25,624,683,1262,1319,614,691,598,681,1266,1319,616,681],
  "time": "2023-09-11T21:14:13.097+0200"
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [6192,457,1132,19768,2524,2545,2497,2536,2524,2515,2503,2554,2489,2529,2511,2521,2511,2564,4820,1273,1312,1275,1304,623,669,1275,665,611,1322,629,672,624,655,1271,1318,1259,1328,1258,1338,615,654,642,656,1265,1333,589,691,612,686,1274,645,637,657,641,667,611,686,602,671,624,680,607,690,621,1303,1276,1322,617,678,1241,1336,1268,1311,1271,652,635,1306,1276,659,623,669,632,1310,1274,656,636,1304,623,681,1265,1323,613,657,643,672,1267,1304,628,659],
  "time": "2023-09-11T21:14:13.219+0200"
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -69,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [7547,1368,7449,11195,2515,2549,2499,2516,2525,2510,2517,2545,2501,2547,2508,2507,2523,2526,4847,1281,1293,1292,1297,634,659,1279,673,613,1308,619,675,617,684,1268,1303,1290,1303,1263,1326,617,674,620,654,1293,1295,638,664,627,670,1254,684,596,697,605,686,615,675,619,679,620,659,616,674,614,1316,1267,1313,643,671,1267,1303,1260,1334,1247,686,590,1329,1287,665,617,675,613,1338,1244,686,628,1293,639,660,1279,1300,638,651,644,652,1269,1325,611,684],
  "time": "2023-09-11T21:14:13.358+0200"
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [10471,17080,2509,2536,2507,2523,2524,2517,2517,2543,2499,2517,2519,2522,2537,2516,4841,1286,1293,1288,1304,629,659,1261,669,632,1310,618,670,637,662,1270,1322,1263,1318,1266,1300,633,660,640,669,1254,1335,619,652,635,657,1273,680,617,677,617,671,615,661,641,670,603,683,608,679,619,1326,1253,1321,614,669,1276,1310,1272,1311,1268,681,632,1290,1288,658,620,670,637,1318,1243,683,614,1316,635,661,1274,1298,641,668,604,683,1258,1319,633,665],
  "time": "2023-09-11T21:14:13.711+0200"
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [7133,1643,6984,11803,2506,2536,2508,2527,2516,2506,2513,2552,2509,2535,2505,2531,2523,2494,4853,1307,1289,1272,1307,623,669,1283,656,615,1321,629,665,622,681,1265,1319,1270,1296,1267,1328,629,659,631,649,1269,1336,618,673,609,685,1269,651,613,681,618,669,637,665,616,674,622,678,616,684,617,1298,1279,1321,613,656,1274,1334,1251,1326,1251,671,634,1311,1267,660,638,666,630,1311,1269,679,614,1308,623,678,1259,1320,609,684,620,693,1231,1328,629,658],
  "time": "2023-09-11T21:14:13.712+0200"
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -71,
  "bits": 56,
[... 101 lines cut ...]
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [8158,19404,2522,2517,2520,2521,2525,2500,2526,2541,2508,2522,2513,2518,2515,2529,4843,1283,1293,1292,1301,631,661,1265,668,633,1312,620,670,636,660,1254,1322,1257,1320,1272,1347,588,687,626,671,1251,1335,597,676,611,684,1267,674,614,687,619,669,608,686,617,674,621,654,640,657,620,1322,1259,1318,635,655,1293,1311,1246,1323,1273,674,615,1311,1268,687,615,673,622,1304,1273,675,611,1341,597,675,1287,1298,633,660,615,670,1279,1317,621,673],
  "time": "2023-09-11T21:14:15.036+0200"
},
{"encKey": 166,
  "address": 14048562,
  "rcode": 2526,
  "command": "Prog",
  "rssi": -69,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [7765,1749,4402,7435,6209,2532,2526,2534,2511,2527,2510,2519,2534,2502,2535,2509,25,2542,2510,2506,4862,1292,1296,1291,1295,639,661,1271,654,635,1307,625,682,608,678,1267,1313,1274,1307,1276,1323,615,674,621,676,1271,1321,612,665,626,669,1258,685,614,668,631,659,609,699,620,652,636,662,636,671,613,1316,1273,1313,619,670,1260,1333,1254,1310,1276,688,603,1323,1264,665,638,671,615,1315,1269,660,616,1343,616,665,1270,1324,614,668,629,659,1283,1295,635,681],
  "time": "2023-09-11T21:14:15.190+0200"
}]

rstrouse, 2023-09-11:
What address do you have set for in ESPSomfy RTS for this shutter?

rstrouse, 2023-09-11:
Did that motor actually jog after only 3 seconds on the remote?

elpigeondindo, 2023-09-11:
yes he does but he always has. It doesn't move though when I do it with the sp.

Here is what I have in my conf. I checked that I had not put it on another shutter.

https://ibb.co/4NwYfVn

elpigeondindo, 2023-09-16:
Please continue to respond

So here's what I did. I set the deviation frequency to 1.58khz.
 my procedure:
- I put the blind on prog
- I press the esp program (it doesn't work)
- I deprog the pane
- add 10khz to the deviation frequency and restart.

I did that up to 180khz but it didn't matter. I don't have a way to see the frequency with software or a device?

elpigeondindo, 2023-09-16:
I just tried adding and changing remotes. everything works normally. It's really strange that the esp can't do it. I think I must not be far from the solution

rstrouse, 2023-09-16:
Yeah this is weird that it will not respond despite the same frames being sent.  Especially, since all the motors are the same and the remotes are the same.

elpigeondindo, 2023-09-18:
I do not really know what to do anymore. Is there no way to read the exact frequency of the remote control? I don't even know if it could help me but hey you never know

rstrouse, 2023-09-18:
Not with a CC1101.  There was another user who figured out that their frequency was 433.38 by using an SDR and sniffing the center.  This could actually be the CC1101 not really sending at the 433.42 center.

elpigeondindo, 2023-09-18:
How can I modify this?

Is there a parameter I can try to see if the blind can finally respond?

rstrouse, 2023-09-18:
That is on the base frequency.

elpigeondindo, 2023-09-18:
frankly, I don't understand anything anymore and I really don't know what to do. They all work except for this one.

rolling code can't help?

rstrouse, 2023-09-18:
I meant the base frequency on the radio setup.  Right now it should be set to 433.42.   It is very strange that this one motor does not work.  You indicated that you reset it to factory settings so I assume it cleared out all the old addresses.  When you cannot get a Somfy remote to respond you have to reset it to factory settings then start over but it sounds like you did that.

The rolling code only has a bearing for the address that is stored in the motor.  If the said address continues to send frames outside of the window of rolling codes stored on the motor it will lock out that address and not allow it to communicate with it.  Factory resetting clears this.

elpigeondindo, 2023-09-18:
I factory reset the engine but I didn't do that to the remote control since it works.

Do you know the method to reset it from the factory?

I will try everything

rstrouse, 2023-09-18:
When you reset the motor did you have to re-pair the remote and set the end stops?

elpigeondindo, 2023-09-18:
yes when I have finished handling the motor, I have 2 minutes to pair the remote control. then I have to do all the end points

rstrouse, 2023-09-20:
Try pairing the ESPSomfy RTS channel first.  There is something very odd about that motor.

elpigeondindo, 2023-09-20:
Instead of synchronizing the remote control, do you want me to press prog on the esp?

rstrouse, 2023-09-20:
Yes after you factory reset the motor press the prog button to pair the remote if that is what you do to get the initial remote paired.  If it is not how you set the initial remote get the latest release and select the virtual remote tab.  This will allow you to send whatever is required for initial pairing.

In fact the virtual remote will let you set the endstops with ESPSomfy RTS and send any other key combinations that may be required.  Keep in mind that some motors also require closing the initial process before adding any other remotes.  It will remain in a limbo state until that happens.  Unfortunately, it is not always the same based on the model of the motor.

rstrouse, 2023-09-24:
@elpigeondindo did the above steps get you sorted out?

elpigeondindo, 2023-09-27:
Sorry for my delay, I've been very busy.

So I tried to reset the shutter but this time by pressing the prog button and not the remote control. The shutter does not move when I press the prog key. Nothing is happening.

elpigeondindo, 2023-09-29:
Finally it's solved!
I bought a new esp32 vroom. I scanned the frequency of my remote control. And you know what?

IT WORKS !

I thank you for your time that you gave me. thank you again very much.

rstrouse, 2023-09-29:
Awesome!  I am going to close this one.  Open a new issue if you have any others (related to ESPSomfy RTS of course).

closed 2023-09-29 by rstrouse