rstrouse/ESPSomfy-RTS #148: Axroll RTS no successful pairing

tremichl, 2023-09-10:
Hi! Thanks for this nice project. It is working fine with some of my Somfy motors. But unfortunately Axroll pairing does not work. 00 in Axroll display is always flashing and no reaction when pressing Prog in ESP-Somfy-RTS. Both Axroll and Software are set to 1-Button mode. Tried also 20 reapeats but no luck. What else can I do?

Michael

rstrouse, 2023-09-10:
Try pairing it after selecting the garage 3-button option.  If it pairs then the Axroll is similar in design to the Dexxo and should be using the Garage (3-button) type.

tremichl, 2023-09-11:
Have set Axroll to 3-button mode and selected the garage 3-button option. No pairing with "Prog" button in pairing dialog. Up-down buttons showed some reaction in Axroll (00 flashing stopped for a while) but also no pairing success.

rstrouse, 2023-09-11:
Hold the pairing button down for at least 2 seconds.  Some of these motors require a medium length press to register the pairing.

tremichl, 2023-09-13:
Sorry, doesn't work either. Have cleared Axroll memory, tried 80bit, 56bit 1-butto and 3-button mode never any reaction pressing "Prog" for more than 2 seconds. Maybe debug outputs from working remotes might help. First from Somfy remote control (with no prog button, pairing with any key) and second form FHEM via CUL.

Thanks

Debug output from Somfy Remote Control pressing sigle button once for start once for stop:
[{"encKey": 164,
  "address": 11424762,
  "rcode": 915,
  "command": "Toggle",
  "rssi": -51,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [102400,102511,102267,32533,2594,2539,2547,2552,2535,2549,2545,2545,2534,2573,2517,2560,2520,2568,2538,2549,2522,2565,2528,2575,2533,2552,2520,2468,4846,1299,1307,1297,1304,675,660,1297,1308,671,651,687,649,1295,1310,1318,658,668,1304,1323,660,641,1324,1308,1297,1311,670,654,1312,667,663,666,643,1302,677,669,1311,643,675,1289,1320,1312,666,662,1308,1297,678,643,1309,694,629,1313,1319,1289,1320,664,670,1297,663,668,1311,1293,687,638,1320,1289,718,613,1335,647,661,1294,693,639,1303,676,659,1300,678,641,1323,659,655,666,656,1317,1293,690,645,673,660,650,671,657,665,669,658,666,645,673,661,674,648,660,678,667,658,667,643,673,662,1299,679,641,1322,658,655,1295],
  "time": "2023-09-13T10:24:02.375+0200"
},
{"encKey": 164,
  "address": 11424762,
  "rcode": 915,
  "command": "Toggle",
  "rssi": -52,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [634,1918,2521,2554,2540,2548,2526,2563,2528,2574,2510,2578,2525,2468,4847,1296,1308,1294,1316,677,641,1298,1308,671,651,687,650,1295,1311,1318,660,669,1303,1325,658,641,1322,1283,1322,1309,667,654,1312,667,662,667,641,1326,663,648,1312,667,656,1314,1296,1318,666,661,1309,1297,677,643,1311,695,629,1311,1320,1290,1319,665,669,1297,663,669,1310,1291,686,648,1319,1291,695,639,1331,651,636,1315,676,663,1301,679,638,1295,1336,669,632,683,642,1301,1329,652,660,670,649,689,648,668,663,665,643,676,664,674,646,660,653,694,633,665,671,671,662,665,642,1306,1314,660,663,668,662,1291,1320],
  "time": "2023-09-13T10:24:02.430+0200"
},
{"encKey": 164,
  "address": 11424762,
  "rcode": 915,
  "command": "Toggle",
  "rssi": -54,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [2569,2539,2546,2546,2544,2528,2576,2532,2555,2521,2568,2538,2445,4861,1291,1294,1309,1318,659,665,1296,1308,672,661,674,645,1313,1298,1311,669,654,1313,1318,657,644,1299,1315,1296,1315,676,637,1305,674,684,646,648,1313,689,633,1313,667,661,1291,1321,1315,649,663,1308,1298,680,642,1307,671,651,1314,1295,1315,1317,662,647,1295,683,643,1328,1300,678,631,1316,1314,667,656,1313,666,662,1281,695,636,1315,680,637,1320,1310,669,658,657,667,1300,1317,656,665,667,634,684,665,651,661,672,648,686,638,669,662,664,668,654,658,672,651,687,650,668,658,1293,1318,667,643,676,661,1298,1307],
  "time": "2023-09-13T10:24:02.577+0200"
},
{"encKey": 164,
  "address": 11424762,
  "rcode": 916,
  "command": "Toggle",
  "rssi": -50,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 24,
  "pulses": [1138,101246,102425,102387,102401,102427,23117,2559,2574,2521,2560,2534,2570,2516,2561,2526,2574,2516,2557,2524,2568,2542,2545,2523,2568,2531,2551,2546,2551,2524,2485,4825,1296,1312,1294,1336,669,640,1315,1299,680,643,678,640,1322,1288,1317,665,673,669,639,1314,674,641,688,644,1304,1317,1312,668,654,664,673,670,638,680,668,658,667,667,656,1312,667,663,1290,1321,1290,693,637,1306,1324,664,647,1313,667,658,1310,1294,1343,1271,693,642,1298,690,644,1314,1316,664,648,1296,1317,717,614,1332,651,663,1301,677,638,1319,660,664,1297,1315,654,666,677,645,659,655,1319,1290,692,634,692,643,673,666,650,669,661,664,669,656,663,648,671,659,666,670,650,659,674,648,686,651,1292,692,644,673,639,1327,1305],
  "time": "2023-09-13T10:24:06.573+0200"
},
{"encKey": 164,
  "address": 11424762,
  "rcode": 916,
  "command": "Toggle",
  "rssi": -52,
  "bits": 80,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [624,1946,2540,2562,2523,2552,2521,2573,2519,2567,2539,2549,2544,2454,4842,1279,1322,1304,1299,688,649,1309,1298,684,640,693,639,1307,1296,1315,672,639,677,645,1309,697,653,637,679,1293,1311,1296,682,644,700,642,673,649,684,625,691,638,683,644,1325,674,649,1286,1320,1315,668,655,1313,1293,685,645,1298,677,645,1309,1324,1311,1292,661,676,1294,686,620,1324,1293,693,639,1306,1300,679,667,1311,669,654,1289,694,633,1312,668,659,1293,1333,652,642,678,659,673,649,1310,1294,688,649,669,660,665,644,673,664,671,645,688,651,666,660,666,643,674,663,673,645,662,664,669,656,1289,695,633,684,646,1324,1293],
  "time": "2023-09-13T10:24:06.628+0200"
}]

Debug output FHEM via CUL:

[{"encKey": 160,
  "address": 11250603,
[... 33 lines cut ...]
},
{"encKey": 160,
  "address": 11250603,
  "rcode": 1904,
  "command": "Down",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses": [3351,2560,2432,2575,2597,2574,2571,2569,2594,2548,2615,2412,2588,2557,5158,1371,1240,1319,1241,574,813,560,627,613,753,600,599,1359,614,612,612,746,1237,1324,633,590,635,727,625,587,630,735,646,555,645,710,1263,1323,1237,622,737,612,608,1369,1228,583,780,1200,770,568,1241,752,614,570,641,745,639,566,643,1323,635,583,1389,607,600,1360,595,628,746,620,1194,773,620,1171,792,594,1222,743,640,566,640,723,641,568,665,1302,642,570,1393,622,584,1374,592,621],
  "time": "2023-09-13T10:36:59.523+0200"
},
{"encKey": 160,
  "address": 11250603,
  "rcode": 1904,
  "command": "Down",
  "rssi": -73,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 13,
[... cut ...] [3343,2558,2610,2379,2631,2535,2595,2571,2588,2566,2581,2407,2629,2509,5208,1321,1235,1360,1230,583,781,598,596,605,776,598,622,1345,618,596,618,729,1243,1344,631,570,658,746,583,611,609,759,627,585,624,734,1222,1355,1218,598,768,593,645,1326,1217,619,763,1206,763,607,1206,756,622,581,623,758,620,574,645,1334,621,597,1370,592,645,1319,644,568,793,569,1237,771,592,1221,741,620,1222,746,620,593,613,721,662,571,642,1337,632,563,1406,578,619,1338,644,575],
  "time": "2023-09-13T10:36:59.696+0200"
}]

rstrouse, 2023-09-14:
I don't see a prog command from the FHEM via CUL.  Is there a prog function?

tremichl, 2023-09-15:
Yes FHEM has a prog function, but I paired it with "on" command because FHEM prog also didn't work with Axroll. This is was FHEM porg does (without pairing):

  [{"encKey": 160,
  "address": 11250603,
  "rcode": 1913,
  "command": "Prog",
  "rssi": -72,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [101880,102407,39216,559,2240,99,25277,35116,474,483,101404,98463,3926,1030,31040,519,481,498,37747,2597,2547,2617,2540,2626,2551,2592,2411,2588,2564,2600,2554,2589,2554,5193,1178,1402,1177,1399,569,641,570,799,568,641,576,792,568,640,571,793,1196,1399,1169,1395,1166,1417,566,643,568,791,1171,1402,1195,640,714,1263,1305,1254,1335,1247,1334,1247,1328,1240,742,632,1174,794,555,659,576,786,584,624,585,775,575,642,580,800,574,1248,1330,1244,1320,1247,1341,1241,716,643,1192,796,568,640,569,790,567,666,548,793,570,638,718,643,24,587],
  "time": "2023-09-15T11:32:55.390+0200"
},
{"encKey": 160,
  "address": 11250603,
  "rcode": 1913,
  "command": "Prog",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [3216,2549,2594,2539,2602,2576,2616,2383,2613,2536,2603,2556,2597,2549,5197,1168,1392,1187,1408,578,625,584,799,572,643,552,795,571,639,575,796,1170,1420,1167,1392,1194,1394,564,642,570,794,1185,1388,1185,646,710,1249,1329,1265,1325,1237,1345,1237,1327,1257,736,617,1184,801,572,644,555,794,570,642,573,795,570,643,572,796,568,1269,1326,1245,1320,1240,1345,1249,721,639,1164,811,563,657,558,801,553,646,581,799,571,641,553,796,572],
  "time": "2023-09-15T11:32:55.496+0200"
},
{"encKey": 160,
  "address": 11250603,
  "rcode": 1913,
  "command": "Prog",
  "rssi": -71,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses": [3214,2555,2588,2564,2604,2557,2621,2373,2619,2546,2593,2574,2595,2547,5016,1355,1394,1170,1416,567,641,573,790,567,641,575,791,569,639,573,794,1152,1419,1193,1378,1199,1389,589,608,604,761,1208,1376,1203,619,729,1242,1348,1239,1323,1247,1347,1243,1321,1259,717,642,1180,788,584,623,563,802,575,642,579,798,552,640,575,796,567,1265,1333,1246,1318,1247,1346,1239,715,640,1193,793,570,651,546,791,572,639,592,792,544,659,568,802,572],
  "time": "2023-09-15T11:32:55.607+0200"
}]

rstrouse, 2023-09-15:
> Yes FHEM has a prog function, but I paired it with "on" command because FHEM prog also didn't work with Axroll.

Well that is a pertinent piece of information.  There is no such thing as an on command on the command by so could capture the FHEM "on" command that you used?

tremichl, 2023-09-15:
Thanks for your time. Because your project is so great I would like to
integrate all my Somfy products. That is the output from FHEM "on" command.
It is the same for pairing and 1-button remote. Even Somfy hardware remote
uses the same button for pairing (and no prog!) and later on for remote
control.

[{"encKey": 160,
  "address": 11250603,
  "rcode": 1914,
  "command": "Down",
  "rssi": -71,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 14,
  "pulses":
[2596,2398,2616,2548,2589,2559,2608,2557,2590,2397,2625,2520,2625,2564,5167,1200,1370,1197,1391,571,641,725,640,592,621,720,654,1171,789,564,663,531,1435,560,651,1342,1245,1331,646,703,639,571,641,719,1263,579,784,561,646,1332,1243,1333,1269,547,794,1196,639,706,1262,567,791,568,647,565,790,567,641,1344,640,568,1396,567,646,1323,662,568,639,721,1243,567,793,1191,641,718,1265,545,794,592,641,569,794,569,646,1321,639,567,1396,592,648,1299,643,591],
  "time": "2023-09-15T17:04:58.125+0200"
},
{"encKey": 160,
  "address": 11250603,
  "rcode": 1914,
  "command": "Down",
  "rssi": -72,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses":
[3365,2396,2596,2572,2568,2566,2605,2555,2600,2541,2443,2570,2593,2545,5200,1187,1405,1182,1397,571,641,704,643,569,641,726,640,1189,793,569,649,541,1417,566,665,1321,1253,1316,638,747,611,608,608,762,1201,610,777,571,622,1357,1224,1345,1241,570,797,1171,651,721,1267,549,791,585,634,563,808,580,617,1333,640,575,1398,572,643,1324,665,569,644,706,1248,570,788,1198,637,721,1241,575,788,585,628,586,776,574,644,1330,646,570,1394,571,641,1329,637,593],
  "time": "2023-09-15T17:04:58.254+0200"
},
{"encKey": 160,
  "address": 11250603,
  "rcode": 1914,
  "command": "Down",
  "rssi": -70,
  "bits": 56,
  "proto": 0,
  "valid": true,
  "sync": 13,
  "pulses":
[3361,2378,2642,2533,2584,2558,2627,2523,2616,2396,2591,2569,2597,2539,5190,1203,1395,1181,1393,566,643,724,636,569,642,725,649,1169,792,593,638,571,1397,567,644,1320,1245,1347,641,568,792,565,639,723,1263,562,786,580,646,1311,1245,1348,1224,593,792,1200,611,744,1244,557,788,587,629,586,777,573,644,1337,645,573,1397,566,644,1324,644,570,640,724,1243,574,787,1194,641,720,1243,566,817,568,625,567,817,567,642,1320,630,596,1395,567,619,1352,631,586],
  "time": "2023-09-15T17:04:58.352+0200"
}]

rstrouse, 2023-09-15:
Ok lets try this since ESPSomfy RTS can and will send this command.  

Step 1: Put the motor into pairing mode
Step 2: Press the Down button on the Edit screen.

[image]

Step 3: After the motor pairs simply open the Pair Shade screen and click the Door Paired button.

tremichl, 2023-09-16:
That worked! Axroll is now paired with ESP-Somfy-RTS. Repeat is set to 3. Can move garage door with down button like 1-button. Link remote is also working. But, because down button now acts as toggle button (up, stopp, down, stopp, up, ...) position information is invalid. Hardware remote doesn't have 3-buttos therefor Axroll is configured for 1-button.

Is it necessary or possible to create a new "Axroll" shade? What would be MQTT topic in that case?

Thanks!

tremichl, 2023-09-16:
Or would it be more straight forward if ESP-Somfy does behave like Hardware remote with one button for pairing, and start stopp?

Debug output from Somfy Remote Control pressing sigle button once for start once for stop:
[{"encKey": 164,
"address": 11424762,
"rcode": 915,
"command": "Toggle",
"rssi": -51,
"bits": 80,
"proto": 0,
"valid": true,
"sync": 24,
"pulses": [102400,102511,102267,32533,2594,2539,2547,2552,2535,2549,2545,2545,2534,2573,2517,2560,2520,2568,2538,2549,2522,2565,2528,2575,2533,2552,2520,2468,4846,1299,1307,1297,1304,675,660,1297,1308,671,651,687,649,1295,1310,1318,658,668,1304,1323,660,641,1324,1308,1297,1311,670,654,1312,667,663,666,643,1302,677,669,1311,643,675,1289,1320,1312,666,662,1308,1297,678,643,1309,694,629,1313,1319,1289,1320,664,670,1297,663,668,1311,1293,687,638,1320,1289,718,613,1335,647,661,1294,693,639,1303,676,659,1300,678,641,1323,659,655,666,656,1317,1293,690,645,673,660,650,671,657,665,669,658,666,645,673,661,674,648,660,678,667,658,667,643,673,662,1299,679,641,1322,658,655,1295],
"time": "2023-09-13T10:24:02.375+0200"
},
{"encKey": 164,
"address": 11424762,
"rcode": 915,
"command": "Toggle",
"rssi": -52,
"bits": 80,
"proto": 0,
"valid": true,
"sync": 13,
"pulses": [634,1918,2521,2554,2540,2548,2526,2563,2528,2574,2510,2578,2525,2468,4847,1296,1308,1294,1316,677,641,1298,1308,671,651,687,650,1295,1311,1318,660,669,1303,1325,658,641,1322,1283,1322,1309,667,654,1312,667,662,667,641,1326,663,648,1312,667,656,1314,1296,1318,666,661,1309,1297,677,643,1311,695,629,1311,1320,1290,1319,665,669,1297,663,669,1310,1291,686,648,1319,1291,695,639,1331,651,636,1315,676,663,1301,679,638,1295,1336,669,632,683,642,1301,1329,652,660,670,649,689,648,668,663,665,643,676,664,674,646,660,653,694,633,665,671,671,662,665,642,1306,1314,660,663,668,662,1291,1320],
"time": "2023-09-13T10:24:02.430+0200"
},
{"encKey": 164,
"address": 11424762,
"rcode": 915,
"command": "Toggle",
"rssi": -54,
"bits": 80,
"proto": 0,
"valid": true,
"sync": 13,
"pulses": [2569,2539,2546,2546,2544,2528,2576,2532,2555,2521,2568,2538,2445,4861,1291,1294,1309,1318,659,665,1296,1308,672,661,674,645,1313,1298,1311,669,654,1313,1318,657,644,1299,1315,1296,1315,676,637,1305,674,684,646,648,1313,689,633,1313,667,661,1291,1321,1315,649,663,1308,1298,680,642,1307,671,651,1314,1295,1315,1317,662,647,1295,683,643,1328,1300,678,631,1316,1314,667,656,1313,666,662,1281,695,636,1315,680,637,1320,1310,669,658,657,667,1300,1317,656,665,667,634,684,665,651,661,672,648,686,638,669,662,664,668,654,658,672,651,687,650,668,658,1293,1318,667,643,676,661,1298,1307],
"time": "2023-09-13T10:24:02.577+0200"
},
{"encKey": 164,
"address": 11424762,
"rcode": 916,
"command": "Toggle",
"rssi": -50,
"bits": 80,
"proto": 0,
"valid": true,
"sync": 24,
"pulses": [1138,101246,102425,102387,102401,102427,23117,2559,2574,2521,2560,2534,2570,2516,2561,2526,2574,2516,2557,2524,2568,2542,2545,2523,2568,2531,2551,2546,2551,2524,2485,4825,1296,1312,1294,1336,669,640,1315,1299,680,643,678,640,1322,1288,1317,665,673,669,639,1314,674,641,688,644,1304,1317,1312,668,654,664,673,670,638,680,668,658,667,667,656,1312,667,663,1290,1321,1290,693,637,1306,1324,664,647,1313,667,658,1310,1294,1343,1271,693,642,1298,690,644,1314,1316,664,648,1296,1317,717,614,1332,651,663,1301,677,638,1319,660,664,1297,1315,654,666,677,645,659,655,1319,1290,692,634,692,643,673,666,650,669,661,664,669,656,663,648,671,659,666,670,650,659,674,648,686,651,1292,692,644,673,639,1327,1305],
"time": "2023-09-13T10:24:06.573+0200"
},
{"encKey": 164,
"address": 11424762,
"rcode": 916,
"command": "Toggle",
"rssi": -52,
"bits": 80,
"proto": 0,
"valid": true,
"sync": 13,
"pulses": [624,1946,2540,2562,2523,2552,2521,2573,2519,2567,2539,2549,2544,2454,4842,1279,1322,1304,1299,688,649,1309,1298,684,640,693,639,1307,1296,1315,672,639,677,645,1309,697,653,637,679,1293,1311,1296,682,644,700,642,673,649,684,625,691,638,683,644,1325,674,649,1286,1320,1315,668,655,1313,1293,685,645,1298,677,645,1309,1324,1311,1292,661,676,1294,686,620,1324,1293,693,639,1306,1300,679,667,1311,669,654,1289,694,633,1312,668,659,1293,1333,652,642,678,659,673,649,1310,1294,688,649,669,660,665,644,673,664,671,645,688,651,666,660,666,643,674,663,673,645,662,664,669,656,1289,695,633,684,646,1324,1293],
"time": "2023-09-13T10:24:06.628+0200"
}]

rstrouse, 2023-09-16:
Now change that shade to Garage 1-button and 80 bits.  It will then start sending the same commands as above.

rstrouse, 2023-09-16:
v2.1.6 contains functions to ignore any toggle repeats from the remote.  This could have caused an issue with the positioning on a Garage door.

The Dexxo motors appear to have the ability to operate as a 3-button once they are paired. (who names these things... Picachu?).

rstrouse, 2023-09-18:
Please download and try this firmware.  I have noticed a variance in the output timing for the toggle command.

https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin

rstrouse, 2023-09-19:
Further on this topic to use the 3-button mode the manual says you have to set the P0 to 04.  I don't really know what that means but it should operate in either mode.

rstrouse, 2023-09-20:
> What will be MQTT topic for "toggle" in 1-button mode?

Actually this is very obscure and I may add some topics in the future but set the following topic `[root]/shades/[id]/direction/set` to 0.

rstrouse, 2023-09-21:
Closing this for now if you have any other issues, create a new one.

closed 2023-09-21 by rstrouse