rstrouse/ESPSomfy-RTS #149: Restore of backup does not restore radio GPIO settings

Calimerorulez, 2023-09-10:
I've just tried to upgrade my customer built m5stack ESP build to 2.1.5. The upgrade failed because of I can't reproduce, so I tried to reflags it with the onboarding firmware file.

I tried to restore the backup and all but the radio GPIO pinout (which is custom in my case because of the m5stack ESP) is restored. Only the default configuration of the GPIO pins, like described in the documentation is set.

rstrouse, 2023-09-10:
Transceiver, Ethernet PHY, and security settings are stored in the ESP32 NVS.  This is because these items are specific to the esp32 and incorrect settings (especially for the ethernet PHY) can destroy the board if they are not set correctly.  Unfortunately, the only way to onboard with the ESP is to flash all of the partitions.  This means the NVS partition is re-initialized with the onboard firmware.

I do have an enhancement that includes the ability to back up the data in the NVS and only restore certain portions of that data in the roadmap.  This has to be done to keep folks from potential disaster.  I have this in my head that if I warn them and they still choose to cook the ethernet chip then I did my part.

rstrouse, 2023-09-12:
I am looking at the potential for this.  I have decided that while I will back up all the transceiver and ethernet settings I will not back up the security, mqtt, and wifi settings.  This presents too many problems.  Also, if the chip ids do not match from the backup I will not restore the ethernet settings or enable the radio.  This will help to make sure users do not toast an ethernet chip.

rstrouse, 2023-09-15:
New functionality coming in the pending 2.1.6 release.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#backup-and-restore

rstrouse, 2023-09-16:
v2.1.6 has been released and contains the functions to backup and restore the radio settings.

closed 2023-09-16 by rstrouse