rstrouse/ESPSomfy-RTS #150: Wrong light setting for Shutters

Mortimerp9, 2023-09-11:
Hello,

Issue:
some shutters show a "light" toggle next to the up/down buttons. This doesn't make sense for shutters.
It looks like when we add new shades, they inherit by default some settings of the previous shade, but if we change shade type (garage to shutter), the new shade inherit settings that are not valid for this type and there is no way to remove that.

Repro:
This is how I think it happened:
I've set a couple of shutters and they did not have a "light" setting (makes sense), then I set up my garage door, and this have a light setting that I toggled on. I then created a new shade, this started the creation form on the garage type, I changed the type to shutter, but this didn't toggle the light setting off. Now for every shade I create, they have a light attached to them.

rstrouse, 2023-09-11:
Go into the configuration and edit the shade.   Choose one of the garage door type then uncheck Has Light.  Now change the type back to the original and press the save button.

It appears that I need to clear that selection when it isn't an option for the type.  Thanks for the report.

rstrouse, 2023-09-24:
@Mortimerp9 I assume you have sorted this out.  Please open another issue if you have any other issues.  I have already added code to clear the flags if the motor does not support the function.

closed 2023-09-24 by rstrouse

Mortimerp9, 2023-09-24:
Thanks for the fix, yes I figured it out, just wanted to flag the bug.