rstrouse/ESPSomfy-RTS #154: WT32-Etho1 and D-Sun CC1101

daviss57, 2023-09-16:
When connecting my WT32-Etho1 to the transevier D-Sun CC1101. Do the terminals labeled on the WT32 correspond with the Somfy GPIO numbers? i.e. IS I012 on the board GPIO-12 on the Somfy RTS dashboard?

[image]

rstrouse, 2023-09-16:
Yes but review the documentation for that board. You must change the settings to match the connection to the dsun.

daviss57, 2023-09-17:
Thanks, I just wasn't 100% on the GPIO labels vs the labels on the board

daviss57, 2023-09-18:
Ok, so i have tried a few times and I have not had any luck getting a signal when I press my remote buttons.

From the below table does it look like i have it wired correctly?

[image]

Jumperm, 2023-09-19:
> Ok, so i have tried a few times and I have not had any luck getting a signal when I press my remote buttons.
> 
> From the below table does it look like i have it wired correctly?
> 
> [image]

Hi,
with the documentation it work out of the box. 
https://github.com/rstrouse/ESPSomfy-RTS/wiki/Simple-ESPSomfy-RTS-device#wt32-eth01-models

daviss57, 2023-09-19:
Thanks.

I've followed to the letter, the only reason I could think for it not
working was the IO pins may not have matched the software GPIO.
Do you know if there is a way to check if the transmitter is
functioning properly?

rstrouse, 2023-09-19:
On the DSUN transceiver side make sure the pins are as they appear.  I have seen a couple of different pinout descriptions.  On the back of the board it should be labeled.  VCC, GND, SCK, MOSI, MISO, GDO2. CSN, and GDO0.

daviss57, 2023-09-19:
Yes, I have double checked the labels to ensure I've wired them correctly.

Not sure where to go from here.

rstrouse, 2023-09-20:
I know this is not what you want to hear but it may simply be a bad board.

wutz-ESP, 2023-09-20:
in your screenshot  WT32-ETH01
SCK - - - - GPIO-15 - - - **GPIO 14**
GDO0 - - - GPIO-19 - - - **GPIO 02**  _TX_
CSN - - - - GPIO-16 - - - **GPIO 12**
GDO2 - - - GPIO-13 - - - **GPIO 35**  _RX_
MISO - - - GPIO-18 - - - **GPIO 04**
MOSI - - - GPIO-14 - - - **GPIO 15**

daviss57, 2023-09-20:
Is the below what I should have

[image]

wutz-ESP, 2023-09-20:
[image]

daviss57, 2023-09-20:
Thanks,

That was before I changed them to match my table. I wasn't sure if the IO2
on the board was the same as GPIO-02 in the software

daviss57, 2023-09-20:
> in this screenshot SCLK is GPIO-15
> 
> [image]
> 
> there you can choose the board [image]

Thanks

I missed that.  Hopefully that sorts it

wutz-ESP, 2023-09-21:
[image]

[image]

rstrouse, 2023-09-24:
@daviss57 did you get this all sorted out?

daviss57, 2023-09-24:
Unfortunately I think I fried the bord when I swapped over to the ethernet
cable. Tried resetting by reinstalling the software but its not emitting a
WIFI signal to log onto it

rstrouse, 2023-09-24:
You have to jumper the firmware write pins on that board to flash the onboard firmware.  Then you must remove the jumper to get it to boot.  Then you have to be very careful to make sure you have wired everything properly or it can break the Ethernet chip.

daviss57, 2023-09-24:
Yep, whatever I did, the little yellow LED no longer lights up/blinks. The
firmware goes through the full write process without issue but nothing
seems to happen when I reboot it.

Have ordered another one. Hopefully I can get that one going.

wutz-ESP, 2023-09-25:
have you seen this instruction?
https://wolf-u.li/flashen-des-wt32-eth01-ein-esp32-mit-lan-wifiwlan/

On first boot you have to wait a long time before wifi getting visible.

[image]

daviss57, 2023-09-28:
Thanks for your help. Recieved by new ESP32 today and have setup. All working 100%. Excelent software. Thank you.

daviss57, 2023-09-28:
Closing now.

closed 2023-09-28 by daviss57