rstrouse/ESPSomfy-RTS #155: no assign IP / keeps transmit wifi signal

jboesh, 2023-09-19:
I have a strange problem when I go to http://192.168.4.1 and select the correct WiFi with password under Network (adapter). then reboot. no IP is assigned. I don't see it as a client in my modem.

do you have any idea what this is about?

The WiFi signal "espsomfy RTS" continues to transmit, but disappears for a while and then comes back on. I don't think this should be the case either?

I have already put new filmware on it but this did not solve it.

rstrouse, 2023-09-19:
This definitely appears to be issues with the WiFi connection.  The socket disconnect and reconnect in HASS confirms this.  Sometimes HASS devs are a bit ridiculous as they throw a warning whenever any client connects to a socket.  Why successfully connecting to a socket is a warning boggles my mind.  Honestly for a platform that is all about integrations to throw warnings just because an integration exists, is like spreading bacon grease all over your body so you don't get wet.  I spent time that I will never get back trying to make that go away.

Lets rule out the settings in ESPSomfy RTS and the router.  First lets make sure we are set up for your WiFi.  I assume by virtue of connection initially you have chosen the SSID and entered a passphrase.  When it connects you should be able to uncheck the DHCP checkbox to see what IP it was assigned, as well as all the other IP information.  Don't leave this unchecked unless you know what you are doing because your router can give that address away to some other device.

Here are some questions to ponder and a couple of suggestions:
1. Are you getting ample signal on the WiFi signal strength as reported from the ESP32 (top of the screen)?
2. Is your Wifi set up as WPA or something else?
3. Do you have special characters in your SSID?
4. Have you disabled 802.11b (This frame causes a lot of traffic during connection on the network for the unlikely chance you may have an 802.11b device).
5. Are you using DHCP?
6. Have you reserved the ESPSomfy RTS device in the static DHCP list in your router?
7. Does your hostname contain spaces or special characters?
8. Turn off SSDP.  Perhaps your network is not dealing with the UPnP packets.
9. Set your timezone and press save.

I hope some of this helps.  Typically if the WiFi disconnects it is because it has lost signal or the lease for the IP address is being dropped from the DHCP server.  You can set up a static IP but the better choice is to reserve the address in your DHCP settings of your router.

If this does not work then we may need to check to ensure the ESP is not simply rebooting.  The most common reason for that is low power.  It will simply brown out and reboot.  If you hook it up to USB and watch the logs through web.esphome.io you can witness whether a reboot occurred.

jboesh, 2023-09-20:
thanks for your detailed response! Yesterday, as you indicated, I reserved the IP address and also set the timezone correctly. I also run Adguard as a DNS server that I temporarily disabled in the modem settings. At some point the esp remained available.

However, because of these problems I have of course tried everything to get it working. including the shutters removed without first unpairing. For example, if I close 1 shutter now, 2 in the house respond. is there any way to solve this? can I simply create a new shutter but with a different name? Or should I clear the memory somehow?

Thanks!

jboesh, 2023-09-20:
I also saw that I could use mqtt.  does this mean that WiFi is no longer necessary?  perhaps because of reliability

rstrouse, 2023-09-20:
WiFi and MQTT are two completely different things.  In fact MQTT is dependent on the connection to the device through TCP which is dependent on Ethernet that is delivered via a cable or WiFi.  The only optional component is MQTT here.  However, if you want to get rid of WiFi you could choose one of the Ethernet wired ESP32s.

Unfortunately, there is no way to clear abandoned addresses from the motors without factory resetting.  However, if you have mistakenly paired a channel to a motor and want to remove it and have a remote controlling a shutter all by itself, you can simply open the memory on that one shutter and send the pair (which is also the unpair) frame to the motor.  It should jog and the address will be removed from its memory.  This means it will no longer hear frames from ESPSomfy RTS.

rstrouse, 2023-09-24:
@jboesh I am going to close this for now.  If you have any other issues please create a new one.

closed 2023-09-24 by rstrouse