rstrouse/ESPSomfy-RTS #156: [feature request] Link external gate status to binary_sensor

ultratoto14, 2023-09-20:
Hi,
I have an exterior gate using a Somfy SGA 6000 motor configured as 3 button garage door in ESPSomfy RTS.
It works perfectly to open and close the door.
I also have 4 keygo remotes using single buttons that open/close the same. 
My problem is that even if I linked the remotes in SomfyRTS, I have false positive statuses in somfy RTS.
I added a magnetic sensor on the gate to have the right status of the door in Home Assistant. It is represented as binary_sensor.

Could it be possible in any way to change the status of ESPSomfy RTS cover and link it to the binary_sensor ?
It could be directly linked or a service to set the current status of SomfyRTS without triggering any RTS command.
What do you think ?
Regards,
U.

rstrouse, 2023-09-20:
That is actually a fine plan.  Let me think about how to organize those endpoints.  I think simply exposing another service on the HA component may do the trick.  The cover entity can be a real PITA but if I extend the services beyond what is stock I may be able to tweak the open and closed state.  I had to do that with tilt only blinds.

rstrouse, 2023-09-22:
I have modeled a couple of services in Home Assistant and you will be able to call a service on the gate to set the current position without moving the gate.  So when the magnetic sensor trips all you need to do is call set_current_position with 0% or 100% and it will reflect opened or closed.

This also goes along with the ability to use external sun and wind sensors by sending the sensor commands to the motor.  All of these will be released in the next couple of days with v2.1.8.

rstrouse, 2023-09-24:
I have released v2.1.8 for ESPSomfy RTS and the HA integration.  You should now be able to set the position with an automation.

ultratoto14, 2023-09-25:
Just tested, it works as expected. Many thanks.

closed 2023-09-25 by ultratoto14