rstrouse/ESPSomfy-RTS #157: Update APPLICATION ESP not responding

Jumperm, 2023-09-21:
Hi,
I update many times my ESPSomfy-RTS. But last two Versionupdates 2.1.X -> 2.1.6 and 2.1.6 -> 2.1.7 fail. I Update first the Firmware and after successful reboot, it shows the new FW and the old Application Version, I update the application. After Uploading my ESP is not responding. Anyone else have such problems?

WT32-ETH01 with CC1101 (E07-M1101D)

rstrouse, 2023-09-22:
I had a problem with a wt32-eth01.  It turned out to be the serial adapter not providing enough power.  It had trouble writing to flash and would brown out.

Jumperm, 2023-09-22:
Clean Flashing work, because I had a 5V Powersupply. Also flashing the ESP via OTA I have the 5V/1A Powersupply.
At the beginning of the week I change from a ESP32 D1 mini to the WT-ETH01. With the D1 Mini I have with one of the last OTA updates the same problem.

rstrouse, 2023-09-22:
Tell me a little bit about your configuration.  I flash firmware and application files quite a bit and never have problems that are not obvious.  It is very odd that the update would not throw an error if it did not complete.  However, the low power condition is the one thing where ESP can think it wrote the files and not actually do it.

If this condition occurs would it be possible to boot it up while looking at the serial logs.  This can be done by web.esphome.io.

Pulpyyyy, 2023-09-24:
Good morning.
I had the same behavior (ESP32 VROOM + E07-M1101D-SMA).
Flashed ino file v2.1.7, backup is done. Everything seems fine.
LitteFS flashed, then esp stops

Flashed with USB + onboard firmware, then restore the backup. Everything works again

rstrouse, 2023-09-24:
It is very odd that the write of the file system fails silently.  I wonder if I am getting some sort of interruption externally.  If this happens again can I get either of you to connect the device up to USB and boot it while watching the logs.  You should be able to do this from the web.esphome.io.  The OTA is supposed to write to the temporary partition and if it fails simply roll back.  Getting some feedback from the log when trying to connect to the device may give some insight into what is failing.

rstrouse, 2023-09-24:
I added some additional checking for the update service to abort when there is a write error.  Unfortunately, these checks don't exist until the firmware and application are updated.  Let's see if v2.1.8 goes smoothly and if not, hook it up to look at the serial log when trying to access the device.

Pulpyyyy, 2023-09-25:
Humm 2.1.8 ends normally. ESP is up and running at the end.
I will try downgrading to 2.1.6 again and flashing to 2.1.7.
Can the 2.1.7 files in the repository be broken?

leonardpitzu, 2023-09-25:
I have the same issue. Device is running a ESP32-PICO-V3 SoC.
`rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0

Startup/Boot....
Preference GFG Free Entries: 469
Preference IP Free Entries: 469
SECURITY   Type:1 Username:[] Password:[] Pin:[xxxx] Permissions:0
Connection Type: 1
NTP Settings 
pool.ntp.org TZ:EET-2EEST,M3.5.0/3,M10.5.0/4
WIFI Settings
 SSID: [xxxxxx] PassPhrase: [xxxxxxxxxxxx]
Mounting File System...
File system mounted successfully

Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
Scanned 6 Networks...
 0:  (-46dBm) CH:6 MAC:8C:85:80:80:05:25
*1: skynet (-50dBm) CH:11 MAC:74:AC:B9:1A:15:1E
 2: phi_F48868C459BB (-55dBm) CH:11 MAC:F6:B1:9C:09:7D:3A
*3: skynet (-67dBm) CH:1 MAC:F4:92:BF:2D:E3:01
*4: skynet (-67dBm) CH:6 MAC:74:AC:B9:C1:DA:16
 5: DIGI-C4Cm (-68dBm) CH:2 MAC:B4:B0:24:97:B9:C1
Connecting to AP
Set hostname to:Shades
**********
Successfully Connected to WiFi!!!!192.168.0.12 (-46dbm)
MDNS Responder Started: serverId=D1D350
App Version:2.1.7
No longer using NVS
Reading header at 0
version:14 len:56 shadeSize:256 shadeRecs:0 groupSize:184 groupRecs: 0 pos:56
Applying radio settings Setting Data Pins RX:8 TX:5
Setting SPI Pins SCK:14 MISO:12 MOSI:13 CSN:15
Radio Pins Configured!
Successfully set up the radio
Enabled receive on Pin #8 Timing: 1
Initializing RX Queue
Successfully connected MQTT client client-000050d3d1988330
MQTT Subscribed to:shades/shades/+/target/set
MQTT Subscribed to:shades/shades/+/tiltTarget/set
MQTT Subscribed to:shades/shades/+/direction/set
MQTT Subscribed to:shades/shades/+/mypos/set
MQTT Subscribed to:shades/shades/+/myTiltPos/set
MQTT Subscribed to:shades/shades/+/sunFlag/set
MQTT Subscribed to:shades/groups/+/direction/set
Loading file /index.html
Loading file /main.css
`

Pulpyyyy, 2023-09-25:
I reproduced it. Here are logs
```
Launching microcom serial connection on /dev/ttyS3.
Press Ctrl+X to exit.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (248) psram: PSRAM ID read error: 0xffffffff

Startup/Boot....
Preference GFG Free Entries: 478
                                Preference IP Free Entries: 478
                                                               SECURITY   Type:1 Username:[] Password:[] Pin:[yyyy] Permissions:0
Connection Type: 1
                  NTP Settings
pool.ntp.org TZ:CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
WIFI Settings
 SSID: [CCCCCCC] PassPhrase: [xxxxxxxxxxxxxxxxxxxxx]
Mounting File System...
File system mounted successfully

Launching web server...
Creating Web MicroServices...
WiFi Mode: 0
Scanned 4 Networks...
 0: AAAAAAA (-72dBm) CH:3 MAC:xxxxxxxxxxxx
*1: CCCCCCC (-73dBm) CH:3 MAC:xxxxxxxxxxxx
 2:  (-81dBm) CH:3 MAC:xxxxxxxxxxxx
 3: Livebox-FAEA (-89dBm) CH:11 MAC:xxxxxxxxxxxx
Connecting to AP
Set hostname to:ESPSomfyRTS
**
Successfully Connected to WiFi!!!!192.168.1.223 (-72dbm)
MDNS Responder Started: serverId=601CEC
                                       Connected to SSDP
SSDP Client Started...
App Version:2.1.7
                 No longer using NVS
Reading header at 0
                   version:14 len:56 shadeSize:256 shadeRecs:7 groupSize:184 groupRecs: 0 pos:56
                                                                                                Applying radio settings Setting Data Pins RX:12 TX:13
                                                                                                                                                     Setting SPI Pins SCK:18 MISO:19 MOSI:23 CSN:5
                Radio Pins Configured!
Successfully set up the radio
Enabled receive on Pin #12 Timing: 1
                                    Initializing RX Queue
Socket [0] Connected from 192.168.1.69 url: /
                                             Loading file /index.html
Loading file /main.css
Loading file /widgets.css```

rstrouse, 2023-09-25:
@Pulpyyyy This was going from 2.1.6 to 2.1.7?  It looks like it never outputs the widgets.css file.  I will attempt the same to see if I can get it to fail.

rstrouse, 2023-09-25:
Thanks for that.  I did this process a few times and it worked fine 90% of the time.  When it failed the script had a cache read failure on the browser.  The files are being served but it was pulling an old version of the script from the cache.  

After I cleared the cache all was good.  Going forward I don't think we will see this again now that I know that the parameter cache needs to be set on all the files which I started doing in v2.1.7.  Unfortunately, the parameters were not set in 2.1.6 and it relied on the reload to clear the cache.

I also went from 2.1.7 to 2.1.8 repeatedly with no failures and it picked up all the script, html, and css on the reload 100% of the time.  I think I will look at potential methods to clear this cache after the update completes so it won't pull the root index,html from the cache.  The trick is to only do this when there is a version change.

Pulpyyyy, 2023-09-26:
I also never failed in upgrading from 2.1.7 to 2.1.8. That's why I thought it was related to the contents of the file. The cache could therefore be the explanation

rstrouse, 2023-10-05:
I am closing this out for now.  I think I can avoid this in the future.

closed 2023-10-05 by rstrouse