rstrouse/ESPSomfy-RTS #159: Grouping Shades does not work - strange behavior

lukekaiwalker, 2023-09-23:
Hi,

I observe a really strange behavior with my grouped shades.

During the process of grouping everything works as expected. The single shades jug when I open the memory and jug again when I pair them to a group.

When I then try to move the shades by clicking the down button in the GUI, completely different shades start moving. They don't have anything to do with the grouped shades.

It seems that everything is getting somehow confused when there is more than one group... Anyhow, I am confused now...

Any idea?

Regards
Luke

rstrouse, 2023-09-24:
Luke upload you backup file here so I can have a look.

However, the way this works is that ESPSomfy RTS is simply sending out a single command for the group.  Those that respond, are the shade motors that have the address of the group stored in their memory.  So in order for a motor to respond it has to have had the address paired to it.   Unfortunately, the prog command is used to both pair and unpair so if you try to pair the address to the motor and it is already paired it will unpair.  Conversely, if you try to unpair an address from a motor it will pair it.

So here is what I would like you to do.  First, open the memory on a motor that should NOT be part of the group.  To do this select a remote channel that controls only that motor.  Then hold down prog button on the back of the remote until it jogs.  If any other motors jog when you do this then these are grouped on that remote channel and you will need to remove power from those motors.

Next click on the Virtual remote tab in ESPSomfy RTS and select the group that the motor should not belong to by selecting the group from the dropdown.  Press the prog button on the screen and that motor should then again jog.  This is an indication that it has been removed from the motor memory.   You can test this by clicking the up or down buttons and the motor should no longer move.

rstrouse, 2023-10-05:
@lukekaiwalker did you get your grouping issues worked out?

lukekaiwalker, 2023-10-06:
Sorry, I was not able to try this yet. I will do that within the next days.
Thanks for your comprehensive answer!

lukekaiwalker, 2023-10-12:
Now I did some tests. First I thougth it works. But meanwhile I am more confused than ever...

I understood that the motors move that have the address of the group in their memory. I created a new group and found out that directly some motors move althoug I did not pair them to the group. I thought that this happens because I paired them once to exactly this group adress because I used different systems in the past and have probably paired a lot.

Then I unpaired all motors from the test group so that nothing happenend when I press a button on the virtual remote. So far, so good... Then I paired three motors to this group again. All motors confirmed pairing by jogging up and down and everything seemed to work. Now only two of the three motors are moving when I press a button on the virtual remote for that group. I don't understand that...

Second test was, to create a new group and give it an address which is far away from the numbers I had so far. I wanted to ensure that this has nothing to do with some tests in the past.

When I do that (and I did it three times with different addresses), some of the motors directly move when I click on the virtual remote, although I did not pair any of them.

Even more strange is that ESPSomfyRTS shows me completely different motors in the overview which indicates that they are apparently paired to that group. However these motors do not move.

I know, all that sounds totally strange. But that is, what happens here. Any clue?

Thanks, Luke

rstrouse, 2023-10-12:
@lukekaiwalker upload your backup file.

lukekaiwalker, 2023-10-12:
Sorry, missed that... Here it is.
[ESPSomfyRTS 2023-10-12T22_51_37.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/12887003/ESPSomfyRTS.2023-10-12T22_51_37.zip)

rstrouse, 2023-10-13:
Thanks for that.  I found an issue where the interface was sending a shade command for the group command.  It was doing this on the Edit Group screen.  However, I sniffed the radio transmissions and it was always sending from group address on the home screen.  I also checked the virtual remote and it moved based on the correct address.  Are you sure you selected the group you wanted to send from the Virtual Remote?

Please install the v2.1.10 firmware and application files from the main repository and let me know if this changes anything.

[SomfyController.ino.esp32.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin)
[SomfyController.littlefs.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin)

lukekaiwalker, 2023-10-13:
Thanks for the files! It looks much better now. I can create groups, pair some motors and they behave like expected.

However, there is one motor which is not working in any group. Pairing works normally. But the motor is not moving afterwards. The shade command still works but no group command moves this motor.

Could it be that the memory of this motor is somehow full? That would mean that I need to reset this motor, right? Or do you have any other idea?

Thanks for all that support!

rstrouse, 2023-10-13:
Yeah that happens when the memory is full.   It will respond as if it received the address but will not store it.  When developing this I had one shade that was used for all the testing and you guess it, it finally stopped working despite acting like it paired.  Resetting it to factory defaults fixed the issue.

rstrouse, 2023-10-28:
@lukekaiwalker I am going to close this for now.  I assume that resetting the motor fixed your issue.

closed 2023-10-28 by rstrouse

lukekaiwalker, 2023-10-29:
Unfortunately, resetting did not fixed my issue... :-(

I was not able to do that earlier. Sorry for getting back to you so late... However, after resetting the motor, I programmed it again and added it as a shade to espSomfyRTS. Then I created a group and added this shade only. When I then move the group nothing happens... I have no explanation. It is the only shade which only works as a single shade and not in any group.

Do you have any idea?

Thanks, Luke

lukekaiwalker, 2023-10-29:
Sorry for my late response. Resetting the motor fixed the issue. Thnaks for your support and this great piece of software...