rstrouse/ESPSomfy-RTS #160: two devices different display

wutz-ESP, 2023-09-25:
Update from 2.1.5 to 2.1.8
[image]
[image]

One device switched to this display
[image]
[image]

closed 2023-09-25 by wutz-ESP

rstrouse, 2023-09-28:
The browser cache needs to be cleared.  It is not picking up the right stylesheet.

wutz-ESP, 2023-09-29:
Sorry, realised after opening this issue. That was the reason closing it immediately.
Dont know how to delete a wrong issue