rstrouse/ESPSomfy-RTS #161: Open and Close not working on my blinds

joerggoetze, 2023-09-27:
Hi, I setup all my blinds as described and at a first glance it seems to be working. My Button and tilt are working, but unfortunately the UP or DOWN commands only tilt my blinds.
I'm not able to open or close it, wether ESPsomfy RTS remote or MQTT or HTTP call.

Might you have a tip what I have been doing wrong ?

rstrouse, 2023-09-27:
Did you select the tilt type as tilt only?

joerggoetze, 2023-09-28:
I tried Tilt Only and all other settings.
I did also lot of tests with different settings (56/80-Bit, all kind of tilt settings, Repeat commands, etc), always with the same result. Unpairing and new installation and update to V2.1.8 results in the same problem.

My remote is a Somfy Telis 4 Modulis with jog dial in the middle, a short arrow press moved my blinds up or down. Long press has no effect.

rstrouse, 2023-09-28:
The motor thinks it is windy so it will not move.  Go into the virtual remote, select the blind from the dropdown and scroll down to the sensor `send` button.  Make sure the windy and sunny checkboxes are unchecked and press send.  It can take up to 2 minutes for the wind condition to clear from the motor.

rstrouse, 2023-09-28:
This also looks like you have your motors set to "euro" mode where a long press lifts and a short press tilts.  To confirm this set the bit length to 80 bits then on the Virtual remote press and hold the up or down buttons to see if the blind lifts.

rstrouse, 2023-09-29:
If you install the firmware and application files from the main repository there is a new tilt type `Euro Tilt` that should work with your setup.

[SomfyController.ino.esp32.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) and [SomfyController.littlefs.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin)

rstrouse, 2023-10-05:
@joerggoetze did you try this out?

joerggoetze, 2023-10-10:
Hello, I was away from home for a few days and unfortunately could not test the new version "Euro Tilt".
Today I installed the update, but unfortunately without success.
I have now restarted the installation from scratch, including the update. Unfortunately it still does not work.
In the meantime I noticed that my receivers are set in US mode, i.e. a short click raises / lowers the blind. 
Also, when I set the position via the slider, the blind goes down in very very small steps one after the other. 

Any Ideas ?
Thx

rstrouse, 2023-10-10:
Did you install the update from the links I provided above?   Please capture the logs when you move the blinds up and down with the remote.  Also, did you set the up and down time for your blinds in milliseconds.  So 10 seconds to go up will be 10 thousand and 10 seconds to go down will be 10 thousand.  The tilt time is also in milliseconds.

joerggoetze, 2023-10-11:
Yep, installed the files you provided, also set the up and down time accordantly to the times they need (41000 ms).
A short log for down/up, stop and tilt appended.
[mylog.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/12873326/mylog.txt)

rstrouse, 2023-10-11:
Can you give me a better idea of what is going on with your shades.  I originally understood that the up and down buttons were nudging the motor.  This happens when the motor is either in the endpoint setting mode or is set to be in Euro mode where the motor nudges with a short press and moves completely with a long press.  Neither appears to be the case here.

Originally you said that the up and down buttons only tilt the slats so that typically means either you have a dual motor (which I am almost certain you do not have based on your remote output) or the motor is set to be tilt only.  That however does not appear to be the case either.

Looking at the log output it appears to be from two separate remote channels.  Is that correct?  Also given what I am seeing from the second remote you appear to have an integrated tilt.  On many of those motors they rotate the slats fairly quickly but you must time this and enter that value as well.  With these types of motors the slats will always move to their full up or down position before moving the blind.

Based on what I see in the log, your blind should be set up just like the pic below except the address should be whatever the address setting you currently have.  It should **NEVER** be 4241664 or 4176128 or any other remote channel address you have.  The two mentioned addresses are the addresses of the two remotes in your output.  ESPSomfy RTS will assign a unique one when you first add the shade record.  If you reuse addresses there will be a constant battle for the soul of the motor.

[image]

joerggoetze, 2023-10-17:
After reading your comments and checking all settings, I thought it could be a good idea to test with a simple Telis remote. So I borrowed one and the remote was showing the same behavior as your software device... now it was clear that my receivers are the problem.
I hesitated to reset my 15 year working setup but at the end the new remote was working and your software as well.

I really appreciate your help and support. Thank you very much.

PS: when pairing a new device the website will not refresh, shaded web page with rotating circle. Refreshing the browser works.

closed 2023-10-28 by joerggoetze