rstrouse/ESPSomfy-RTS #162: Awning: Wind Sensor Command Does Not Work - Virtual Remote

BillyFKidney, 2023-09-28:
Attempts to send the `wind` command result in no action from the awning.
[image]

There is a wind sensor on our awning but I would like to be able to use the virtual remote to send the command as well.
[image]

Did I miss a configuration setting / setup step?

rstrouse, 2023-09-28:
If you have a wind sensor then yes you should link it like a remote.  If you do not then sending a Windy command should have put a warning symbol over the icon and disabled the remote.

BillyFKidney, 2023-09-29:
That answers what is wrong with the icon:
[image]

I am going to try and get the Wind Sensor added now…just have to figure out how to trick it to think it is attached to the awning….while shaking it in my hand. 

The instructions were not clear on the linking the Wind Sensor as a second remote (but logically it makes sense now). Perhaps as an enhancement the Wind Sensor check box is disabled until a second remote is connected and designated as the Wind Sensor?

rstrouse, 2023-09-29:
Actually, if you do not have a sensor then you can emulate it with a weather app.

rstrouse, 2023-09-29:
If it is already up then it cannot retract anymore.

BillyFKidney, 2023-09-29:
> If it is already up then it cannot retract anymore.

Awning was fully extended from ESPSomfy RTS.

rstrouse, 2023-09-29:
In your screenshot it is retracted.

BillyFKidney, 2023-09-29:
Awning this moment:
[image]

Button being pressed:
[image]

No action from the awning motor.

[image]

rstrouse, 2023-09-29:
Since you have a wind sensor why are you trying to emulate one?  Did you pair your physical sensor with the motor?

rstrouse, 2023-09-29:
What ESPSomfy RTS is doing is sending the windy frame to the motor.  It is simply acting like the Eolis.

BillyFKidney, 2023-09-29:
I have the physical wind sensor working with the motor and ESPSomfy monitoring it.
This is the current status in ESPSomfy:
[image]
[image]

I will abandon using the virtual wind sensor.

However, the status in Home Assistant appears backwards.
[image]
[image]
[image]

Is this due to the hardware wind sensor…or a different issue? (Thanks for so much help by the way)

rstrouse, 2023-09-29:
It is due to the way the cover domain works in HA.  It is as correct as it can be because if you switch it the icons will be wrong and it will disable the wrong one.  If you flip the position in ESPSomfy RTS the icons will be wrong and the text will reflect the other way.  It is maddeningly stupid and despite complaints about it the devs insist it is correct.

BillyFKidney, 2023-09-29:
Hmm, I will have to look at creating a template sensor to show the position in a different way.

Or perhaps HA should evolve the cover entity to better support awnings (Open = Extended, Closed = Retracted).

rstrouse, 2023-09-29:
Yeah I don’t understand the resistance.  Bear in mind the limits are not in core the issue is with the ui project.

closed 2023-09-29 by BillyFKidney