rstrouse/ESPSomfy-RTS #163: Fav position on press 'my'

elpigeondindo, 2023-10-02:
Hi !

I would like to know if there is a solution to make the 'my' button work directly.

Let me explain, if I go up or down it works. when it goes up or down I can stop it with the 'my' button.

But I can't put the shutter in favorite position by just pressing 'my'.

A parameter to change?

rstrouse, 2023-10-02:
Click and hold the my button then set it.  From then on out, when the motor is at rest it should go to the my position when you press it.

elpigeondindo, 2023-10-02:
Thank you for your answer.

I press my for a long time, the window opens, I put 95% for example and press set my position.
but nothing happens. if I understood correctly he should go down to the position I indicate to him and go jogging but nothing happens.

a weird thing too and it has kept 95% of the my in memory but when I press the key it stops halfway. odd.

rstrouse, 2023-10-02:
Yes it should what motors do you have?

fjrodrigo, 2023-10-21:
Hi, first of all, congratulations for this great and amazing job. I have installed your firmware in a ESP32-WROOM-32 with a E07-M1101D transceiver and the my button doesn't work but up and down work right. I have observed that I'm not able to see any log in the Log screen and I think that's my issue, the RX doesn't work, but I've tried all the pins without any different result. Please, do you have any idea or how can I solve the log issue ??? Thank you very much in advance

rstrouse, 2023-10-21:
Please open a new issue for this.  The issue discussed here has to do with the stored position on the motor.  We will be troubleshooting your transceiver.

rstrouse, 2023-10-28:
@elpigeondindo did you follow the steps I outlined above and did it fix your issue?

elpigeondindo, 2023-11-06:
Sorry for the delay, I did lots of testing.
I used groups but I think there is a bug. I will try to explain it

I create the GR1 group. I add my shutter 1 then 2..3 and finally 4.
Everything is going well, the shutter is moving and I can add them. When I test the group only shutter 1,2 and 3 work but not 4. I then delete shutter 4 from the group. It's just shutter 1-2-3, okay? but when I want to test the group, it works, shutter 1, 2, 3 and 4 work. But in the interface, shutter 4 does not change state

I don't know if I can make you see the problem

elpigeondindo, 2023-11-06:
> @elpigeondindo did you follow the steps I outlined above and did it fix your issue?

Yes it works.

closed 2023-11-06 by elpigeondindo