rstrouse/ESPSomfy-RTS #164: Reordering does not work on touch devices

devinfluencer, 2023-10-04:
Hi,
on my Android 11 phone with Edge browser I can't reorder shades with drag and drop. Instead the list scrolls when I try to drag an item.
The same on a touch laptop.
Using the mouse works fine.
Solution might be to use two-finger drag for scrolling or to display a "move mode" checkbox on touch devices.
Thanks,
Thomas

rstrouse, 2023-10-04:
Tap and hold.  The element will begin to float then you can move it.

devinfluencer, 2023-10-04:
Android: No, the item is grayed out and when I move down the 'reloading' turning arrow appears. The grayed out state can only be reset by leaving the settings page.
Windows (Edge): Tap and hold has no other effect than scrolling.

rstrouse, 2023-10-05:
In the next release this should be fixed.  It turns out that the touch screens have disabled the HTML5 drag and drop so I needed to perform some manual work.  The browser will do some detection then if the touch is held for 1 second it will change to a dragger.

devinfluencer, 2023-10-06:
Super. I'm interested in the details as I have another web app using dnd in development. Can you give a hint where in the sources we are?

rstrouse, 2023-10-06:
If you look at the code in index.js you will find that the answer is to revert back to pre-HTML5 processing and handle it all manually based on touch events.  You will find it around line #2246 on the setListDraggable method.  The key is that the press-n-hold needs to pause/wait for a period of time to switch modes or the scroll functions will overtake it.

rstrouse, 2023-10-12:
v2.1.9 has been published that fixes this issue.

closed 2023-10-12 by rstrouse