rstrouse/ESPSomfy-RTS #165: Access to unused GPIO

radiorx49, 2023-10-04:
Hello
Thank you for paying attention to my request.
I control my Velux shutters with ESPHome using a homeassitant switch which controls a GPIO. I connected each button of a Velux remote control to a GPIO (one for up, one for down). The homeasssitant switch controls the power supply to the GPIO for 500ms which closes the remote control button. (Simulation of pressing the remote control button)
You can watch
https://esphome.io/components/switch/gpio.html
for the code that inspired me. There is no limit switch and the duration is not necessary as it can be set in homeassistant.
In fact, you just need access to the unused GPIO via a switch exposed in homeassistant to be able to use them as you want through HomeAssistant (control of a relay for a garage door, a lamp, simulation of pressing a button ...).
This is the code I use in ESPHome:
switch:
  - platform: gpio
    pin: 12
    name: "Velux Up"
    id: VeluxUp
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: VeluxUp
  - platform: gpio
    pin: 13
    name: "Velux Dwn"
    id: VeluxDwn
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: VeluxDwn
  - platform: gpio
    pin: 14
    name: "Velux Stop"
    id: VeluxStop
    on_turn_on:
    - delay: 500ms
    - switch.turn_off: VeluxStop
cover:
  - platform: template
    name: "Velux Esph"
    open_action:
      - switch.turn_on: VeluxUp
    close_action:
      - switch.turn_on: VeluxDwn
    stop_action:
      - switch.turn_on: VeluxStop
    optimistic: true
    assumed_state: true

Thank you for your attention and at your disposal for tests if necessary.
Eric

rstrouse, 2023-10-04:
How many relays do you have?  Is pin 14 actually attached to a relay?

radiorx49, 2023-10-04:
Hello

 

I don’t drive the remote command with a relay but with a optocoupler.

 

You can see information in the pdf file (in french)  or

 

mrflobow/velux_remote: ESPHome Config for Velux Remote + Wemos D1 Mini (github.com) <https://github.com/mrflobow/velux_remote> 

 

I use a Wemos D1 mini for my velux covers. Pin 12 for up, 13 for down and 14 for Stop. The stop is not obligatory.

But any output GPIO of the ESP32 will be OK. The GPIO can be used directly if we need a direct voltage or can drive a relay via a transistor, a led directly or simulate a push button with a optocoupler.

 

at your disposal for further information if necessary

Thanks

 

Eric

rstrouse, 2023-10-04:
I see.  A stop relay is really not needed.  The only reason for that would be if you were connecting to another remote and these were inputs.  The way this will work is when moving up the up pin will be pulled high and the down pin will be pulled low.  Then when moving down the down pin will be pulled high and the up pin will be pulled low.  When at rest neither pin will be pulled high.

I assume the motors only have forward and reverse.

radiorx49, 2023-10-05:
Good morning

Yes the relay is not needed.

 

In the circuit to control the up button for example, the GPIO is set high for 500ms. The optocoupler LED is lit and drives the optocoupler transistor, which brings the Up button into contact with ground. We therefore have a simulation of pressing the button.

It's the same thing for the down GPIO (button).

 

The goal is just to be able to control the GPIOs by putting them at high level for a time.

 

Via the switch (linked to the GPIO) that you expose in homeassistant you can set the delay (for 500ms to control a garage door or a shutter for example or for 10 minutes to turn on a light via a relay or a LED directly).

Have a good day.

Eric

rstrouse, 2023-10-05:
I added GPIO motor support so please try this out and let me know how it is working.  This should operate your GPIO shades just like they were hooked up to a radio so pull the firmware and application binaries from the main repository and update your device.

[Firmware](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin)
[Application](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin)

After you have updated go to the shade definition that is for GPIO and select GPIO as the protocol.  This should then allow you to select the UP and DOWN gpios.  The up pin will be pulled high when the shade is moving up and the down will be pulled low.  It will reverse when going down.  Stop will pull both pins low.  Let me know how it works.

rstrouse, 2023-10-06:
Here is a quick writeup on configuration.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Controlling-Motors-with-GPIO

radiorx49, 2023-10-06:
Hello

 

I quickly looked at your messages. I'm busy this weekend, I am at work.

I will test the new features as soon as possible and let you know.

 

Thanks for the work.

 

Eric

 

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Controlling-Motors-with-GPIO

radiorx49, 2023-10-07:
Hello

 

Thanks another time for your very good job.

I made some test with the 2.1.9 firmware and application.

 

The GPIO are working OK except for the dry contact.

 

But some remarks :

 

The GPIO are working OK with all type of shade you choose but I don’t like the fact that the GPIO is at a high level as long as the shade is going up or down. For exemple, the velux shade is going down in 29 seconds, value that i put in the up and downtime  text box. So the GPIO is at high level during all the moving time (so it consumes battery and I don’t know if the components of the remote command like this). The command for velux shade need only a pulse on the remote button to go down or up. I try to put 100ms in the up and down time and it’s enough to command the shade (so GPIO at high level for 100ms). The shade stop alone with the end stop in the shade motor. In this case, The 100ms are enough to command the shade but the icon of the position of the shade is not good as the complete moving time of the shade is 29sec.

A solution I can propose (but I am not a good programmer and I don’t know the procedure you use in your program) is to separate the time of the GPIO is high and the time of the shade is moving. If then you choose GPIO You can add, a check box and a text box to choose if you want a continuous high level of the GPIO (the duration of the up or down time as actually) or a pulse time and the time in the new text box. With this, the time of the GPIO is high is not linked with the moving  time of the shade (the icon position is OK) and you can command all type of shades, continuously pressed button ou pulse button. Easy to say …

Proposition : adding  third GPIO to connected to MY button (stop button).

 

If I choose Dry contact, nothing happen. I don’t have oscilloscope but only a voltmeter so I can’t see if the GPIO is going high for a very short moment or no. I try to connect my remote command (same connection as the shade test) and nothing happen. So if the GPIO is going high, the duration is low than 100ms (I can see on the voltmeter the GPIO going high If I choose 1 button garage with 100 ms time) and not enough to simulate a press on the  button of the remote command.

Another proposition to correct this, is that you have to make  the same fonction as the 1 garage button actually but without time and with only one GPIO. A press on the button put the GPIO high, a new press on the button put the same GPIO down. Or also a time box where you can put  0 to have the GPIO high until you press the button a new time or put how many time you want the GPIO high and after this time, the GPIO going down automatically. This will be a good way to command light, irrigation ou most other things (I don’t know if you want to limit your program to shutters management).

 

These are my observations.

It's functional but with some possible arrangements that will make your program (it is already excellent) better.

 

Thanks for your attention and always at your disposal for new tests if you want.

Excuse me for writing English( I am more comfortable in French), I hope it’s understandable.

 

Have a good week end.

 

Eric

radiorx49, 2023-10-07:
A photo of the test

rstrouse, 2023-10-07:
Unfortunately, your pictures and video never made it to your post so I am completely in the dark.  Are you controlling a remote with the GPIO or are you operating relays for a motor?

radiorx49, 2023-10-07:
Ok I am not AT home i Can t send  you again thé picture. (It s only a picture of  connections.)I don t sent you vidéo.
I control a remote with 3 Gpio. One for the UP Burton of thé remote one for down Button and one for stop Button.

⁣F1ISJ Eric​

rstrouse, 2023-10-07:
Just to be sure I understand.  It sounds like you dissected a remote and hooked it up to GPIOs.  Hence the reason for the momentary presses.

rstrouse, 2023-10-07:
Alright I think I have figured out what you are doing.  You have torn apart an existing IO-HomeControl remote and are triggering button presses on that remote.  I originally thought we were talking about a bare naked motor that had no radio control on it.  I had it in my head that since there were not limit switches that you must be dealing with nude motors.  My bad but now ESPSomfy RTS supports this configuration.

However, I also added support for your configuration.   In this configuration ESPSomfy RTS will provide momentary commands through the GPIO to the buttons on your remote.  This also means that you must define the pin for the my button.

Please update your firmware and application from the main repository.  This has not been released yet as I need to enlist you as my test subject.   After you update the application binary please hard refresh your browser.  This is critical since you have already installed the pre-release of the application files.  If you are running on a windows system CTRL+F5 in the browser will do the trick but Apple products have no such function so you have to clear your browser cache.

After you have cleared the browser cache, select `Remote` from the protocol dropdown and the GPIO for My will appear.  Fill this in with the gpio that you have attached to the my button on that remote and press Save Shade.  The duration of the button presses are defined by the number of repeats.  The larger the number of repeats the longer the button will remain pressed.  Repeats from the Virtual Remote also work with this.

[image]

The main issue with this implementation is that there is no feedback from the IO remotes.  So if you use another remote the positions will be messed up.  Sadly, HA exacerbates this issue with the ridiculous button disabling crap that is part of the UI.  It's cute but maddening.  However, if you use RTS remotes and link them to a channel, then pitch your IO remotes over the fence into your neighbors yard you will have control from a remote and the positions will be retained.

When I get a spare moment I will update the docs for this.

radiorx49, 2023-10-09:
Hello

 

Here is my test result. I put it in table format, it’s simplier.

 

Remote Roller

 

during repeat time (pulse)

 

 

 

 

Up button

Dwn button

Stop Button

 

 

 

GPIO Up

High

L

L

 

 

 

GPIO Dwn

Low

H

L

 

 

 

GPIO Stop

L

L

H

 

 

All is working OK for what I have to do. A command for roller that need a pulse pressed button.

 

 

Remote garage 1 button

 

during reapeat time (pulse)

 

 

 

 

1st click

2nd click

3rd click

4th click

5th click

 

GPIO Up

L

L

L

L

L

 

GPIO Dwn

L

L

L

L

L

 

GPIO Stop

H

H

H

H

H

You can remove the GPIO Up and Down. Only the Stop GPIO is changing state every time you click the button (High for a pulse / low / High for a pulse, low …)

 

 

Remote Dry contact

 

during repeat time (pulse)

 

 

 

 

1st click

2nd click

3rd click

4th click

5th click

 

GPIO Up

L

L

L

L

L

 

GPIO Dwn

L

L

L

L

L

 

GPIO Stop

H

H

H

H

H

Same behavior as above, same remark (remove 2 GPIO)

 

 

Relay Roller

 

during Up and Dwn time

 

 

 

 

Up button

Dwn button

 

 

 

 

GPIO Up

H

L

 

 

 

 

GPIO Dwn

L

H

 

 

 

It’s seem OK for me if you want to drive a roller that need a continuous press on a up or down button (perhaps the necessity to have a stop button (without GPIO linked) that put the 2 GPIO (up and down LOW)

 

 

Relay Garage 1 button

 

during Up and Dwn time

 

 

 

 

1st click

2nd click

3rd click

4th click

5th click

 

GPIO Up

H

L

L

L

H

 

GPIO Dwn

L

L

H

L

L

Also OK for me if you want to drive a roller that need continous pressed button with a stop (Up, stop, down, stop, up, stop, down ……)

 

 

Relay Dry contact

 

 

 

 

 

 

 

 

1st click

2nd click

3rd click

4th click

5th click

 

GPIO Up

L

L

L

L

L

 

GPIO Dwn

L

L

L

L

L

Not working. All the GPIO are always low.

 

I think to have a complete set of possibility, in this case (Relay / Dry contact) , there are 2 possibilities

you can put only 1 GPIO drived by one button. Each time tou press the button, the GPIO is changing state (toogle button)

or 

2 buttons, one for ON, one for OFF (or STOP) . ON button pressed, the GPIO is High, OFF (or STOP) button pressed, the same GPIO is going Low.

To have this behavior : continually On/ continually off / continually on / continually off / continually on … 

That is the behavior we are waiting for driving a relay.

Relay Dry contact

 

 

 

 

 

 

 

 

1st click or ON button

2nd click or STOP button

3rd click or ON button

4th click or STOP button

5th click

 

GPIO Up

H

L

H

L

H

 

You have done a great job again. THANKS.

Always at your disposal for further information or test

 

Eric

radiorx49, 2023-10-09:
For information, 

1 time repeat here is enough to put the remote command ON (Velux rollers). I put 2 in the box to be sure.

 

Good evening.

 

Eric

rstrouse, 2023-10-10:
I think the formatting of your table didn't work.  I'm not sure what worked and what didn't.  Perhaps this is because you are posting via e-mail.

rstrouse, 2023-10-10:
I have tried to put this together from the formatting of your post.  It looks like everything is working except the 1-button options for Relay and Remote.  Is that correct?

radiorx49, 2023-10-10:
Hello

 

This is the result in pdf in the file

You are right. All is working except the mode relay/dry contact

Eric

radiorx49, 2023-10-11:
> Unfortunately, the pdf is stripped when you respond via e-mail. However, I posted a fix for the 1-button garage door controller and dry contact.

Hello 

 

Sorry but i will be away from home for 5 days so can’t test the new version.

Did you have a direct mail without passing via github, I can send you the pdf before going this afernoon. ?

 

Eric

rstrouse, 2023-10-12:
I have gone through all the scenarios, updated the GPIO wiki, and published v2.1.9.  I have it all covered now for all the shade types.  It also supports dry contact and 1-button garage doors correctly now.

rstrouse, 2023-10-28:
I am going to close this for now.  I believe we have all the bases covered with the release of v2.1.9 and v2.2.0.

closed 2023-10-28 by rstrouse