rstrouse/ESPSomfy-RTS #166: Thousands of "Name has no usable address" in logs

billmoseley, 2023-10-04:
I had a power failure at my house and when Home Assistant came back up my logs were filled with the messages below.  When I disable ESPSomfy-RTS integration they stop (and re-start when reenabled):

```
2023-10-04 09:08:44.995 ERROR (SyncWorker_9) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.001 ERROR (SyncWorker_8) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.007 ERROR (SyncWorker_0) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.012 ERROR (SyncWorker_7) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.017 ERROR (SyncWorker_6) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.023 ERROR (SyncWorker_4) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.028 ERROR (SyncWorker_5) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.033 ERROR (SyncWorker_2) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.038 ERROR (SyncWorker_2) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.044 ERROR (SyncWorker_7) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.049 ERROR (SyncWorker_6) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.056 ERROR (SyncWorker_6) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.061 ERROR (SyncWorker_8) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.066 ERROR (SyncWorker_0) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.072 ERROR (SyncWorker_3) [websocket] [Errno -5] Name has no usable address - goodbye
2023-10-04 09:08:45.077 ERROR (SyncWorker_9) [websocket] [Errno -5] Name has no usable address - goodbye
```

You can see how frequently this is happening. There's no other lines in the home-assistant.log file that provide more info (e.g. a stack trace or a log message indicating what really failed), so maybe something uncaught?

Enabling debug logging only adds:
```
2023-10-04 09:26:20.928 DEBUG (MainThread) [custom_components.espsomfy_rts.controller] Manually updated espsomfy_rts data
2023-10-04 09:26:20.930 DEBUG (MainThread) [custom_components.espsomfy_rts.controller] Manually updated espsomfy_rts data
```

My guess is what triggered this is my ESP32 appears to be not functioning now. Seems unlikely related to the power failure but power cycling the device has not brought it back.

```
$ ping espsomfyrts.local
ping: cannot resolve espsomfyrts.local: Unknown host
```
But the issue here seems that this isn't being caught and results in filling the logs.

I'm running:
ESP Somfy RTS: 2.1.8

Home Assistant 2023.9.3
Supervisor 2023.09.2
Operating System 10.5
Frontend 20230911.0 - latest

rstrouse, 2023-10-04:
This looks like there is no response from the MDNS.  Reserve the IP address in your router DHCP reservations.  In HA edit the hub configuration and set it to the IP address not the MDNS name.

The message is coming from the websocket library.  This also is probably not from ESPSomfy RTS as this will not try to reconnect after failure for a minimum of 20 seconds.  The message about manual updates is interesting as this comes from the update coordinator in HA.  Perhaps shut down your HA instance completely and restart it.

I will review all the caught exceptions to see if there are more to be caught but this doesn't trigger like this in the test environment even if I shut down (unplug) one of the devices.

billmoseley, 2023-10-04:
Yes, I've now used a static IP for the device and added it to the HA config and all is well. Thanks.

Something odd was causing it to retry -- and retry often! I'll leave this open if you want to look, but my issues seems resolved.

```
➜  fgrep goodbye log.save | head -1
2023-10-03 20:53:48.092 ERROR (SyncWorker_0) [websocket] [Errno -5] Name has no usable address - goodbye
➜  fgrep goodbye log.save | tail -1
2023-10-03 20:58:37.012 ERROR (SyncWorker_12) [websocket] [Errno -5] Name has no usable address - goodbye
➜  fgrep goodbye log.save | wc -l
27990
```

rstrouse, 2023-10-05:
I am going to close this one.  I have not been able to duplicate this no matter how hard I try.  I think those messages may have come from the MDNS service when it was trying to resolve the name in HA but I have added some exception catches just in case.

closed 2023-10-05 by rstrouse