rstrouse/ESPSomfy-RTS #167: Moving shades are flooding network with MQTT messages

tremichl, 2023-10-04:
Hi again! 
Have enabled MQTT and read messages with FHEM MQTT2 broker. Moving shades causes a FHEM crash. Investigating the problem showed up that hundreds of MQTT messages in a few seconds were sent (direction, position, target, mypos, sunSensor, windy) during moving shades.  

Reducing MQTT messages during moving for topic position only and once or 2 times per second (or one per percent) should solve that issue. What do you think?

Receiving LWT (Online, Offline) and IP address via MQTT would also be appreciated.

Thanks again for this nice project!

rstrouse, 2023-10-05:
I would not have thought that any number anywhere close to that would bring down a broker.  I have several projects that send way more data than that and it doesn't begin to yawn at it.   Are you sure the subscribe is crashing the FHEM MQTT2 broker and not something downstream of that.  Either way I will look at ways to reduce the traffic.  

I cannot keep last sent so I may need to create some setters that do a publish on change instead of a global state change.  Then only send individual ones whenever they change.  This means that each move will emit 100 publishes throughout the transition from fully open to closed.  However, the direction and target would only emit when they change.

Do you need to subscribe to every topic?  Are you only reflecting the state position?  If you are not using any of the other topics you should not subscribe to them if it creates a load on FHEM.

I will have to take a look at the ESP32 LWT for PubSubClient.  It's not overly sophisticated so I may need to do some digging into it.

rstrouse, 2023-10-05:
If you install the firmware and application from the main repository (not the current release) it will only publish MQTT topics on the edge changes.  It also implements a status LWT that indicates online and offline.

[Firmware](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin)
[Application](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin)

Download these two files and update your device.  Let me know how it goes.

tremichl, 2023-10-06:
Big improvement. Thank you! 

In case of of a short or fast moving roller, for example under 10 seconds, there are still 10 or more messages per second. Maybe one message every 5% or 10% would be enough. Just a suggestion. For me it is fine now as it is.

Thanks again for your great work and fast responding.

rstrouse, 2023-10-06:
That socket should be able to handle waaaay more datagrams than that.  It should not bring down the broker even with a high QoS.  I suspect the handler is performing some synchronous process that gets in the way of the incoming messages or the processor that is processing the messages is performing expensive operations in a synchronous manner like screen refreshes or writing to permanent storage.

I am pretty sure that removing the unnecessary traffic and processing state changes on the edge case will solve the problem all around since there are no longer multiple state values being changed in the process thread.  The typical MQTT broker is capable of processing several hundred thousand datagrams per second and an ESP32 can get nowhere close to that about a cycle of 100ms or so is about where this will land.  I am going to close this for now as I don't think even with very fast movements there will be issues since the emitter and the processor on the ESP32 loop is a limiting factor.

closed 2023-10-06 by rstrouse

wutz-ESP, 2023-10-16:
@tremichl - i am using ESPSomfy-RTS in fhem mqtt2 and http://[IP-ESPSomfy-RTS]:8081/shadeCommand?shadeId=1&get=shades 300. Mqtt2 sometimes did not get right position until opening ESPSomfy-RTS webinterface.
Websocket would be great, but i am not able getting readings like mqtt2
https://forum.fhem.de/index.php?topic=135247.msg1288976#msg1288976