rstrouse/ESPSomfy-RTS #170: View use of remotes in HA

jboesh, 2023-10-07:
Is it possible that I can view the use of the remotes in HA?  then I can use that information in automations.  so not only of the shades but the remote of the shades

rstrouse, 2023-10-07:
Yes attributes on the entity indicate which remote address was used and whether that was an internal command to ESPSomfy RTS or a standalone remote.

last_cmd: last command sent
cmd_source: remote or internal
cmd_address: address used for the command
cmd_fired: unix time when the command was fired in milliseconds

[image]

jboesh, 2023-10-08:
it looks like I dont have such attributes, can I enable all of them somehow?
[image]

May I ask what internal as cmd source means? Is this through the internal ip / web server?
[image]

rstrouse, 2023-10-08:
Cmd source is where the command came from.  For instance, remote or internal where remote means the command (up, down, stop, ...etc) was picked up from the radio.  Internal means that ESPSomfy RTS sent it out on its radio.

Has a remote or ESPSomfy RTS been used to move that shade since the integration was started?

jboesh, 2023-10-09:
Now that I use the shutter via the remote, I can see the attributes. Will they remain there until a restart of HA, for example?

rstrouse, 2023-10-09:
The remote address will always be the address you assigned to the ESPSomfy RTS shade.

rstrouse, 2023-10-09:
The reason why the attributes disappear is because this data is only kept when the command event is fired in HA and is dropped when the entity is re-initialized.  This is just how HA works and there is no way around this.

closed 2023-10-10 by jboesh