rstrouse/ESPSomfy-RTS #171: Direct FOTA Update

011V32, 2023-10-09:
Hello,
would be great if if you could extend the update mechanism.
A button in the UI to check for updates (or even check automatically) would be great.
And a (second) button to update the firmware and application direct from the repo without downloading it to the PC.
Additionally to update the esp also with the HA update so that the two applications match each other...
Thx.

011V32, 2023-10-18:
I don't know how, but RaspberryMatic did also the FW Update of the CCU over HomeAssistant:
[image]

rstrouse, 2023-10-18:
It is not that it cannot be done but it will need to be crafted from whole cloth as the data returned from github is massive in terms of size.  There is one library that does it but the dependencies do not match and I run out of PSP.  This is not a raspberry pi with an operating system and warehouses of system resources.  It is an MCU with many fewer resources.

So it is a slow slog through the data which I have already done.  I will do my best to get this function added.  At the very least it will check for the latest release.

rstrouse, 2023-10-23:
I will be releasing this functionality next weekend.

rstrouse, 2023-10-28:
Version v2.2.0 now includes a github update function to download and install the firmware and application from the repository.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Updating-ESPSomfy-RTS#github-update-method

closed 2023-10-28 by rstrouse