rstrouse/ESPSomfy-RTS #172: Host Name setting isn't being applied

hugepants, 2023-10-13:
Hi, thanks for your great work on this project.

The Host Name setting (defaults to `ESPSomfyRTS`) in `System`>`Options` doesn't appear to be used. On my OpenWrt router, a DHCP request comes in from host name `esp32-974B68` and I can access the web UI at `http://esp32-974B68.lan` but not `http://espsomfyrts.lan`. It's currently using part of the MAC address in the host name.

I've tried versions 2.1.6 and 2.1.9 - same issue on both.

rstrouse, 2023-10-13:
This is actually two distinctly different issues.  mDNS and DHCP in ESP32 has me pulling my hair out.  It is strange that it responds with esp32-[chipid] sometimes for the hostname in DHCP.   So I did some research and it appears that I am not the only one that has experienced this issue.  After some digging and it appears that most have had to reset the connection before setting the hostname, then set the hostname, and then connect.  I posted a change that does just that.

Please pull the [SomfyController.ino.esp32.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.ino.esp32.bin) and the [SomfyController.littlefs.bin](https://github.com/rstrouse/ESPSomfy-RTS/blob/main/SomfyController.littlefs.bin) from the main repo.  Then flash your device with these.  I have reset the connection before connecting so it does not pick up what is cached from the ESP32 when it obtains an IP address.

The DCHP router list will keep the lease cached so it will not change in that list for a while.  However, if you ping it should find it at ESPSomfyRTS and not the esp32-[chipid] which is embedded in the chip.

hugepants, 2023-10-13:
Thanks for the quick response and fix!

I've updated, but the hostname is still showing as `esp32-[chipid]` - even after I clear the DHCP lease cache from the router, but perhaps it is still cached in some other way.

Interestingly, I can't ping `ESPSomfyRTS.lan` but I can ping `ESPSomfyRTS.local`:

```
$ ping ESPSomfyRTS.lan
ping: ESPSomfyRTS.lan: Name or service not known

$ ping ESPSomfyRTS.local
PING ESPSomfyRTS.local (192.168.1.177) 56(84) bytes of data.
64 bytes from esp32-974B68.lan (192.168.1.177): icmp_seq=1 ttl=255 time=3.36 ms

$ ping esp32-974B68.lan
PING esp32-974B68.lan (192.168.1.177) 56(84) bytes of data.
64 bytes from esp32-974B68.lan (192.168.1.177): icmp_seq=1 ttl=255 time=13.3 ms
```
On my router the local domain is set to the OpenWrt default `.lan` (in dnsmasq).

ESPHome defaults to using `.local`, but [it can be changed with this `domain` setting](https://esphome.io/components/wifi.html) so I have it set to `.lan` on my ESPHome devices. Could this have something to do with it?

rstrouse, 2023-10-13:
Hmmm... I am not sure why they (ESPHome) are doing that as zeroConf and avahi use .local.   The esp_idf also appends .local to the mDNS responder service but I will take a look to see what is in there.  Perhaps it can be rewritten but that will also mean that the zeroConf resolution will need a rewrite.  

It is very odd that your router is resolving esp32-974B68.lan since the ESP32 never advertises that host so is there a DHCP record for that?  At least as of the .10 update.  So this has to be cached somehow on host resolution or it is a DHCP reservation.

rstrouse, 2023-10-13:
Alright after searching and searching, I have determined that the .local domain is hardcoded into the idf and there is no way to change the top level domain.  Zeroconf is also reliant on that to determine whether it is an inside or outside address in mDNS.

rstrouse, 2023-10-13:
I think I know the mystery.  If you look at the .local ping your router is redirecting to the `esp32-974B68.lan` host.   I'll bet if you set a dhcp reservation with the proper host name it will respond to ESPSomfyRTS.lan (do not include .lan in the hostname).   However, the top level domain of .local is the RFC reserved domain for local network addresses from the spec and will always respond from mDNS that way.  The tld of .lan is being resolved by your router using a local DNS service (not mDNS) so it does not look outside your local network.

I am not sure what ESPHome is doing with this unless they are assuming a proper DNS resolution exists on the router.  .lan is expected to route externally through DNS first then come back inside but only if there is a local DNS service.

hugepants, 2023-10-14:
> It is very odd that your router is resolving esp32-974B68.lan since the ESP32 never advertises that host so is there a DHCP record for that? At least as of the .10 update. So this has to be cached somehow on host resolution or it is a DHCP reservation.

I've cleared all traces of cached devices in the router and there is no DHCP reservation set for it, however it's still getting the `esp32*` hostname through DHCP (on FW and App version 2.1.10)

```
Sat Oct 14 09:57:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) b0:b2:1c:97:4b:68
Sat Oct 14 09:57:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.1.177 b0:b2:1c:97:4b:68
Sat Oct 14 09:57:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.177 b0:b2:1c:97:4b:68
Sat Oct 14 09:57:02 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.177 b0:b2:1c:97:4b:68 esp32-974B68
```
I must have missed something, so later on I will replace the router with a backup unit which the ESP has never connected to which should produce a fresh DHCP lease.

> Alright after searching and searching, I have determined that the .local domain is hardcoded into the idf and there is no way to change the top level domain.

Thanks for looking into it, that's good to know.

> I think I know the mystery.

Aha! So if I understand correctly, the local router DNS has an entry for the device at `*.lan` and the mDNS name is `*.local`. Both are valid and accessible concurrently.

>  I'll bet if you set a dhcp reservation with the proper host name it will respond to ESPSomfyRTS.lan

This will most certainly work (I'll try it later), however it shouldn't be necessary if the ESP device is advertising the `ESPSomfyRTS` host name in the DHCP request instead of `esp32*`, so hopefully this will be the case with the replacement router.

rstrouse, 2023-10-28:
@hugepants I am going to close this for now.  I believe both the .lan and .local addresses should be valid.  If adding the DHCP with the proper host name gives you grief let me know.  mDNS on the ESP32 is a bit green and requires a bit of voodoo to get it working reliably.

closed 2023-10-28 by rstrouse