rstrouse/ESPSomfy-RTS #176: After it lose wifi connection does not control the blinds anymore

jbrandek, 2023-10-20:
Hi thanks for your work on this really nice project and crazy all the documentation, thanks for taking the time to explain.

I did the first device and it has been working perfect, then i made 4 more for the rooms, and i can pair them and it works for a few days. 

What i notice is that when a device lose the conectivity to the network and rhe reconnect it does not control the blinds anymore and i have to repair it. I thought it was that they had bad wifi, but if tou press the remote it sync the status in home assistant. Just lose the ability to control the blinds.

What can it be that is losing the control of the blinds? Any chance also that one device intefearing with the other?

rstrouse, 2023-10-20:
> What i notice is that when a device lose the conectivity to the network and rhe reconnect it does not control the blinds anymore and i have to repair it.

Do you mean that you have to go through the pairing process where you use the prog button on the remote?  This should never need to be done unless the motor loses the address stored in it.  Is this 1 device or all 4 of them?

Interference between two devices is also highly unlikely.  So let's rule this one out.  Lets also take Home Assistant out of the picture by checking to see if you can reach your device web pages when it fails to control the blinds.

If you have access to at least two of these devices.  Open the radio log for one device and send a command using the other and then try it the other way.

jbrandek, 2023-10-20:
yes the whole process again, if i dont do that i cannot control them any more...
yes i also tried on the device webpage i can connect to it and se it but when i send the command nothing happens
yes i can see them in the logs if i send a command in other device

jbrandek, 2023-10-20:
> yes the whole process again, if i dont do that i cannot control them any more...
> yes i also tried on the device webpage i can connect to it and se it but when i send the command nothing happens
> yes i can see them in the logs if i send a command in other device

> Do you mean that you have to go through the pairing process where you use the prog button on the remote? This should never need to be done unless the motor loses the address stored in it. Is this 1 [...]

yes the whole process again, if i dont do that i cannot control them any more...
yes i also tried on the device webpage i can connect to it and se it but when i send the command nothing happens
yes i can see them in the logs if i send a command in other device

rstrouse, 2023-10-20:
Ok so this has nothing to do with losing wifi connection.  What is odd is that your motors seem to drop the address from it.  Is this a particular motor or all of them?

jbrandek, 2023-10-20:
> Ok so this has nothing to do with losing wifi connection. What is odd is that your motors seem to drop the address from it. Is this a particular motor or all of them?
> 
> EDIT: Please upload a backup of your configuration.

All of them, yes its really strange
[ESPSomfyRTS 2023-10-20T22_51_41.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/13057212/ESPSomfyRTS.2023-10-20T22_51_41.zip)

rstrouse, 2023-10-20:
I assume that between your devices you do not share any addresses.  Is that correct.  If every address is not unique any command from any duplicated address will cause the motor to not respond.

jbrandek, 2023-10-22:
What do you mean by shared address? Each device control each room, the motor respond if i do in in the remote from somfy. Do you want me to upload a backup from each device?

jbrandek, 2023-10-22:
> I assume that between your devices you do not share any addresses. Is that correct. If every address is not unique any command from any duplicated address will cause the motor to not respond.

ok i see the address yes they are some with same address, im changing that and i will try with that, do i have to link them again or just by changing the address they should work?

rstrouse, 2023-10-22:
Every time a motor hears a duplicated address is will cause all other remotes on that address to stop working for the number of commands used by the other remote.  This will continue and seem as if it loses the remote.  Evenually you will need to factory reset the motors so that it can start over with unique addresses.

rstrouse, 2023-10-28:
@jbrandek did your issues vaporize when you assigned unique addresses?

jbrandek, 2023-10-29:
Hi thanks i was trying that on the weekend i think so, they are all working again thanks for the help i close the issue

closed 2023-10-29 by jbrandek