rstrouse/ESPSomfy-RTS #177: Logs window doesn't show information and "my" button doesn't provoke any blind movement or stop it

fjrodrigo, 2023-10-21:
Hi, first of all, congratulations for this great and amazing job. I have installed your firmware in a ESP32-WROOM-32 with a E07-M1101D transceiver and the "my" button doesn't work, but "up" and "down" buttons work as expected. I've tested in 6 different motorised blinds with RTS Telio remote control (9 years old). With "my" button I'm able to record a new position, mantaining "my" button pressed and selecting "SET MY POSITION", but after that only the remote control takes the blind to that position, "my" button doesn't provoke any movement or action in the blind. I have also observed that I'm not able to see any log in the Log screen and for all that I think that's my issue, the RX doesn't work, but I've tried all the GPIO pins (GPIO02, GPIO04, GPIO12, GPIO13, GPIO14, GPIO16 and GPIO17) without any different result. If I move the blind from the main page, the window picture doesn't show the percentage neather the blue blind picture moving up or down as you explain in the documentation. I can bring if needed the serial traces with the Arduino IDE environment. Please, do you have any idea or how can I solve the log issue ??? Thank you very much in advance

rstrouse, 2023-10-21:
Just so you know the logs do not show any transmit data from the radio.  This is by design because we only have confirmed information from the radio on the receive side.  On the transmit side there is no verification so that would just be noise in the log.  If the radio not initialized message goes away then we have to assume that the radio actually sent the command.

If you are not seeing any responses in that log from your Telis then the most common problems are listed below in the order of how often I see them.

1.  Wiring does not match the configuration.  This often happens because folks are familiar with RS232 serial comms and reverse the RX and TX lines to the transceiver.   The CC1101 can receive on either GDO(0 or 2) pin but can only send on GDO0.  So TX needs to be connected to the GPIO pin that is labeled GDO0.  On the E07-M1101D transceiver this is pin #3.  Please make sure TX is connected there.  The RX setting can be the same so if you connected pin #3 on the transceiver to GPIO13 on the ESP32 then you can set both the RX and TX pins in the configuration to that same pin and not connect GDO2.
2. Incorrect power connections.  There have been a couple of times that folks connected the 5v pin from the ESP32 to the transceiver.  This will result in disaster as transceiver can be damaged.  Its a crap shoot as to whether this happens but it will eventually fail if connected to 5v.  As well wiring the either power pin to the ground on the transceiver will cause it to fail.
3. Poor connections with Dupont connectors.  If you are like me and have a drawer full of Dupont connectors they eventually get loose after a surprisingly short number of connect and disconnect times.  A poor connection on any of them can cause the radio to malfunction.
4. There are integrations out there that require you to clone a remote and as a result the integration and the remote compete for rolling codes.  Despite my best efforts I cannot seem to stop people from repeating this.  The address of a shade or group in ESPSomfy-RTS should never be the same as a physical remote.

Once you get the receive working so that it reliably hears your remote then link the remote to shade you defined in ESPSomfy-RTS using the link remote button.  It should then make the shade in ESPSomfy-RTS respond appropriately to the commands from your remote.  Only then can ESPSomfy-RTS show the position when you control the motor with the remote.

fjrodrigo, 2023-10-22:
Dear rstrousse, I really followed your instructions, but the result has been the same. In fact, I have two transceivers and two ESP32 boards and I used them to run an example from library SmartRC-CC1101-Driver-Lib v2.5.7 (New_Method_without_GDO), configuring one ESP32 as transmitter and the other one as receiver and it works. However, if I try with the example Old_Method_with_GDO it doesn't work. Could you find any relation between this information and the issue I have?

rstrouse, 2023-10-22:
This really seems to indicate a wiring problem.  Please take some pics specifically showing which pins on transceiver are connected to which pins on the ESP32.  

Then shoot a screenshot of the radio configuration screen as well.

The example from the driver old method repo will never work with decoding Somfy signals.  That is because the modulation does not match and the preambles are not fixed.  If you did not make changes to it, the default GDO0 pin is actually GPIO6 so the old method example will never work if you wire it based on the example.  The new method will also not work to actually decode the signal because of the packet differences.

The pin settings from the default library are as follows.  The diagrams also depict an older version of the CC1101 breakout so the pin numbers are different.  

| CC1101 | Desc |  ESP32 |
|---------|--------|-----|
| 1 | GND | GND |
| 2 | VCC | 3v3 |
| 3 | GDO0 | GPIO2 |
| 4 | CSN | GPIO5 |
| 5 | SCK | GPIO18 |
| 6 | MOSI | GPIO23 |
| 7 | MISO | GPIO19 |
| 8 | GDO2 | GPIO4 |

fjrodrigo, 2023-10-22:
I attach you the [pictures...

[image]
[image]
[image]
[image]
[image]
[image]

fjrodrigo, 2023-10-22:
Sorry, I missed the remote picture... 
[image]

rstrouse, 2023-10-22:
Ok I have verified the wiring from your pictures and it is correct.  

| Color | CC1101 | Desc | ESP32 |
| ------ | ------ | ------- | ------- |
| Black | 1 | GND | GND  |
| Red | 2 | VCC | 3v3 |
| Yellow | 3 | GDO0-TX | GPIO2 |
| Orange | 4 | CSN | GPIO5 |
| Purple | 5 | SCK | GPIO18 |
| Blue | 6 | MOSI | GPIO23 |
| Green | 7 | MISO | GPIO19 |
| Brown | 8 | GDO2-RX | GPIO4 |

Click on the scan frequency button, then hold down one of the buttons on your remote.  Are you getting any signals?  If it hears the remote it will capture the frequency and the RSS in the box provided.

[image]

fjrodrigo, 2023-10-23:
I followed your instructions, but the result is always the same with the two different ESP32 boards I have. In addition I have soldered one of them to a PCB taking into account the experience with Dupond cables in issue #3, but again I got the same, I mean, it doesn't hear the remote...
[image]
[image]
[image]
[image]

rstrouse, 2023-10-23:
On that board you have soldered up do not change any wiring and change the settings on the radio page to this.  Then press save radio.  After that try again.  If it does not work then there is something wrong with that transceiver.

[image]

rstrouse, 2023-10-23:
Uggh.  I didn't even see that in the screenshots until just now.  The socket is required from the web client to communicate with the ESP32 which in turn communicates with the CC1101.   I didn't think you could even get to that screen without a connection.

I have mine hooked up through Home Assistant.  That links them to HomeKit and ultimately Siri.  I don't use Alexa but I can't see it being too hard once you have a connector.

rstrouse, 2023-10-26:
I am going to close this for now as it appears you were in AP mode.  I will remove the options in the future to eliminate that confusion.

fjrodrigo, 2023-10-26:
Ok, that's would be perfect for next people!. Thank you so much for your support. I'm now focused to hook up my ESPsomfy RTS device through Home Assistant and link all to Alexa. The problem is that the only way I've found to translate all the blinds functionality to Alexa is by opening my netwok to allow Alexa connect for outside with my Home Assistant, and that is not trusted for me...

rstrouse, 2023-10-26:
Yeah everything for Alexa has to run through their cloud.

closed 2023-10-26 by rstrouse