rstrouse/ESPSomfy-RTS #178: Can't reveice signals from somfy remote

sapanica, 2023-10-23:
Hi there,

After your last help, which is awsome, the conclusion was that it may be a bad receiver.

So last week the new receiver has arrived, but it still hasn't fix my problems.

I made this issue, so i won't mess up your huiden repository.

Any suggestions? Replace the cables between receiver and the esp32?

rstrouse, 2023-10-23:
If I remember correctly here are the things we established.

1, You have an E07-M1101D.
2. You have an ESP32 NodeMCU from AZDelivery
3. Your remote is a Telis 4

So lets go through a few things up front.  Look closely at your remote and make sure it is not an IO remote.  I do not think it is but look for the letters IO or Pure IO.   Next verify that the leds when you press a button glow red not green.

If all of that is correct then take some pics of the connections on the transceiver and connections made to the ESP32.

sapanica, 2023-10-24:
Hi there,

The lights on the remote are red when a button is pushed

Could you check the pictures on the repository, or should i upload them again?

rstrouse, 2023-10-25:
Upload them here

rstrouse, 2023-10-28:
@sapanica I have dug up your pics from the home assistant community.  If your wiring on the transceiver is still this.

[image]

Did you verify that you have these wired to the ESP32 like below?

[image]

Also you must perform the initial network connection before it can start receiving signals.  So you should not be connecting to 192.168.4.1 when looking for frames from the remote.

sapanica, 2023-10-29:
Sorry that i haven't replied earlier, i've been very busy lately. I'll try to check everything tonight and let you know!

rstrouse, 2023-10-29:
No worries at least it wont get lost in other replies here.

stanojevicha, 2023-11-12:
Hi, sorry for hijacking this issue but have same problem.
Below is the link to items I use and how they are conected.
But in general ESP looks like working fine, I have tried two different radios, no software error also but there is not any event on LOG screen.

I have placed a picture of mines remote because they have IO in their name.
As you mantion it seems lika that can be issue but are the option in that case?

[image]
[image]
[image]
[image]
[image]
https://www.aliexpress.com/item/1005005626327947.html
https://www.aliexpress.com/item/1005002603656188.html
https://www.aliexpress.com/item/1005002074380868.html

rstrouse, 2023-11-12:
@stanojevicha I regret to inform you that this solution will not work for you.  You do not have 433Mhz motors.  You have IO which using 868Mhz and an entirely different protocol.

stanojevicha, 2023-11-12:
Thanks for info.

elcriuz, 2023-11-14:
Same problem here.
Board: ESP32-WROOM-32
D-SUN CC1101

Checked wiring a hundred times now – I am pretty confident that the pin-layout is fine. How likely is it that the radio is faulty? I have only this one.

rstrouse, 2023-11-15:
This is a silent gliss remote.  Did you set your base frequency to 433.92?  After you do that check the log.  I suspect this will probably emit RTV protocol.  It will show as 56V for the bitness in the log.

elcriuz, 2023-11-15:
Yes, I set the base frequency accordingly. Nothing shows up in the logs when I click on the remote. The logs stay empty no matter what I try. This is what makes me curious.

rstrouse, 2023-11-15:
Just to be sure you used the wiring diagram for the D-SUN and not the E07-MC1101D.  I have had more issues with the D-SUN transceivers.

I am not positive but I believe there have been other users with the Silent Gliss.

elcriuz, 2023-11-15:
Yes, I am pretty sure I used the correct wiring diagram. There is also a pin description printed on the board which matches the settings in ESPSomfy.

rstrouse, 2023-11-15:
@elcriuz try increasing the RX bandwidth about 50k at a time pressing save radio in between.  Give it at least 10 seconds between each to allow the radio to reset.  I have had a few dead D-SUN transceivers.

jcdel09, 2023-11-22:
hello ,
modify the connections, I had problems with this configuration. For me it works with this ports on tx/rx.
[image]

elcriuz, 2023-11-25:
Just tried all the mentioned above plus a new E07-M1101D in addition to my D-SUN and a new ESP32 (ESP32-WROOM-32). I checked the pins a million times. Still nothing in the logs. I am really lost here. Anything else I could try?

rstrouse, 2023-11-25:
@elcriuz just to be sure you have set the base frequency to 433.920MHz... correct?

elcriuz, 2023-11-25:
Yes, see screenshot attached.

Lg,
Christoph

jcdel09, 2023-11-26:
do you have another ESP32 and ,or M1101? in case it is a broken equipment ?

elcriuz, 2023-11-26:
I have several other ESP32 (same brand - AZDelivery) but only the sone M1101. I had the same issue with my D-SUN receiver thought.
So its either:

• I am really stupid and I am missing something obvious in my setup
• I have really bad luck with my components
• ESPSomfy just can’t see my Silent Gliss remote. Unfortunately I don’t have any other 433Mhz remotes I could try at the moment

rstrouse, 2023-11-26:
Unfortunately, I don't have any Silent Gliss equipment.  Although, there was another user with equipment from Silent Gliss, it is possible that the protocol varies within the equipment line.  You can test the equipment by simply creating two devices.  ESPSomfy-RTS will listen to another ESPSomfy-RTS device quite readily.  If the receiver is working then either device will show the output of each other.

domdomson, 2023-11-27:
Hi all,

first of all: Thanks for this project.

My shutters are the only reason why i can´t migrate from fhem to home assistant.

As for testing i set-up two esp32 with cc1101 each.

When i send something from the on instance i get a log entry on the other instance. Nice.

Unfortunatly i don´t recieve anything from my remote.

I´m using shutters from nobily (they should be made by dooya, i think), After a little bit of digging i found that the signal should consist of 40 bits.

https://www.nobily.de/rolladentechnik/rolladen-sets/rolladen-profi-sets/funk-mechanisch/nobily-profi-set-pr5-inkl.-funk-timer-exquisit-handsender

Is there a way to get espsomfy running with it?

Currently fhem is sending something like this:
2023.11.27 18:43:13 4: Dooya set value = D_Rollo1D stop
2023.11.27 18:43:13 4: Dooya_sendCommand: D_Rollo1D -> message :P16#0001100000000111100000101011000101010101#R10: 

Regards,

Dominik

closed 2024-02-04 by rstrouse