rstrouse/ESPSomfy-RTS #179: Blinds within apple home app

Jack123uk, 2023-10-23:
Hi, 
Im having some issues with the function of the  blinds when I have imported them into the home app. Everything works great up and down etc, but let’s say I open the blind 37% it says it’s closing them, and vice versa, once the operation is complete the state is always correct, it’s just getting confused with the wording within the icon if that makes sense! It’s not actually getting the up and down wrong, it’s just when I press to open the blinds it will state (closing..). I’m not sure if this is a limitation of the home app, or if there is something I can change. Any help would be great. 
Thanks.

rstrouse, 2023-10-26:
I'll double check this but it may come down to limits with the cover entity in Home Assistant.  It uses percentage of open for some types and percentage of closed for others.  Which shade type do you have selected?

rstrouse, 2023-10-26:
I checked most of the iterations of the shade types and they are reporting opening and closing properly in Home Assistant how do you have this configured?

rstrouse, 2023-10-28:
I checked this against the Home Assistant Homekit plugin and it appears you are correct.  However, this is correct in home assistant and if you switch it, it is incorrect in home assistant.  It just shows on the homekit app that it is closing when it is opening and vice versa.   What a goofball thing.  Home Assistant is showing it correctly.

I think there is an issue with the HomeKit bridge or the HomeKit app itself.

Jack123uk, 2023-10-29:
Sorry I’ve been away for a few days, thank you for checking this for me. So if im using the blinds more in the home app then home assistant I could change the setting correct? And it should report the correct states? Which setting would I change for this? 
Thank you again and what a great integration you have developed.

rstrouse, 2023-10-29:
I don't think there are any settings on Home Assistant or in ESPSomfy RTS that can affect this.  Honestly, it appears to be a problem in HomeKit.   It's very odd that it reports exactly opposite on the app but only during movement.

Jack123uk, 2023-10-29:
I agree it’s very odd! But that’s Apple for you,  they make everything difficult or awkward! 
Not to worry thank you for looking into this for me, at least i know it’s not an issue on my side! Thanks

rstrouse, 2023-10-29:
To be fair Home Assistant is braindead when it comes to the position and the devs are very sensitive about being told its wrong.  For instance, the percentage used is in percentage of open.  So a fully closed blind is 0% regardless of whether you tell it that 100% is closed.  

Maybe that is logical for a window or an awning but it makes little sense for blinds, shutters, shades, curtains... etc.   Oddly, you can get it to say closing when going to 100% but once it's at 100% it will then report that it is open and it's infuriating.  So perhaps this is not as much an Apple issue as it is a Homekit Bridge issue.

rstrouse, 2024-01-09:
I am going to close this one out as I have done all I can do to make it report properly.

closed 2024-01-09 by rstrouse