rstrouse/ESPSomfy-RTS #181: Only down

abvlm, 2023-10-25:
Hi,

It's sometime since I was using this amazing project, but some days ago it started to act weirdly.

Currently I am only able to put down the shades with ESPSomfy, never up or use the My button. 

If someone have an idea from where it comes from. Also I am on version 2.1.7 and tried to update over the air but was not able having an error.
[image]

rstrouse, 2023-10-25:
Please reserve your ip address in your router with dhcp.  Then use the ip address.  There are some limits with mDNS on the ESP.  Some networks do not consistently resolve the name on all ports.  I am looking at alternatives that do not require constant processing on the ESP.

abvlm, 2023-10-25:
It already has an assigned ip address and I only access it with its IP and not the hostname.

rstrouse, 2023-10-25:
Can you check to see what is in the browser log.  It looks like it cannot resolve the host.

abvlm, 2023-10-26:
It does not seems to be the case
[image]

Is there a way to backup my config by serial connection ? (I prefer not going throught reconfiguring all shades).

Thanks

abvlm, 2023-10-26:
After monitoring the serial log, here is what I see :
`Enabled receive on Pin #12 Timing: 0
                                    Sending Shade Command
Received:{"shadeId":3,"command":"down"}
Getting Next Rolling code 1
                           CMD:Down ADDR:651452 RCODE:1 REPEAT:0
Enabled receive on Pin #12 Timing: 0
                                    Sending Shade Command
Received:{"shadeId":3,"command":"up"}
Getting Next Rolling code 1
                           CMD:Up ADDR:651452 RCODE:1 REPEAT:0
Enabled receive on Pin #12 Timing: 0
                                    ./components/esp_littlefs/src/littlefs/lfs.c:1886:debug: Bad block at 0x1
./components/esp_littlefs/src/littlefs/lfs.c:1892:warn: Superblock 0x1 has become unwritable
./components/esp_littlefs/src/littlefs/lfs.c:1886:debug: Bad block at 0x1
./components/esp_littlefs/src/littlefs/lfs.c:1892:warn: Superblock 0x1 has become unwritable
./components/esp_littlefs/src/littlefs/lfs.c:584:error: No more free space 433
Timing Somfy: 360ms
                   ./components/esp_littlefs/src/littlefs/lfs.c:1886:debug: Bad block at 0x1
./components/esp_littlefs/src/littlefs/lfs.c:1892:warn: Superblock 0x1 has become unwritable
./components/esp_littlefs/src/littlefs/lfs.c:1886:debug: Bad block at 0x1
./components/esp_littlefs/src/littlefs/lfs.c:1892:warn: Superblock 0x1 has become unwritable
./components/esp_littlefs/src/littlefs/lfs.c:584:error: No more free space 433
Timing Somfy: 360ms`

rstrouse, 2023-10-26:
That looks like the flash on that device has either failed or the partitions are marked as unreadable.  Which version of the ESP are you using?

Also run this in a browser to get the configuration from memory.  `http://<IP ADDRESS>/shades.cfg`

abvlm, 2023-10-26:
I am using a ESP32 WROOM.

I managed to flash again the ESP32 but had to disconnect the radio pins (due to strapping pin were used [https://www.esp32.com/viewtopic.php?t=5970](according to this). And before flashing use :
`esptool.py write_flash_status --non-volatile 0`

How can I restore the shades.cfg file ?

rstrouse, 2023-10-26:
Are you using the recommended pin settings from the documentation?  Did you use the original SomfyController.onboard.esp32.bin file?

You should be able to simply copy that to a file named ESPSomfyRTS.backup and restore that file directly.

rstrouse, 2023-10-28:
@abvlm are you back in business?

abvlm, 2023-10-28:
> Are you using the recommended pin settings from the documentation? Did you use the original SomfyController.onboard.esp32.bin file?
> 
> You should be able to simply copy that to a file named ESPSomfyRTS.backup and restore that file directly.

Yes I am using the recommended pins with the released file.

> @abvlm are you back in business?

Yes everything is back to business. Thanks a lot for your help. But I would be interested in some documentation on the backup file (I need to tweak it a little)

rstrouse, 2023-10-28:
Ok so here goes.  The backup file is a fixed length record for all the shades and groups.  Followed by a single line each for the settings, network, then radio.  I have included a full output of the data in the file but you should be able to do all of this through the interface including setting rolling codes if you must.

The first line of the backup file is a fixed length header separated by commas.  It's layout is as follows.

| Name |  Length | Comments
| ------ | ------ | ------ |
| Version | 3 | This is the version of the backup file |
| Header Length | 3 | The total length of the header |
| Shade Rec Length | 5 | The total length of a single shade record |
| Shade Rec Count | 3 | The number of shade records in the file |
| Group Rec Length | 5 | The total length of a single group record |
| Group Rec Count | 3 | The number of group records in the file |
| Settings Rec Length | 5 | The total length of the settings record |
| Network Rec Length | 5 | The total length of the network record |
| Radio Rec Length | 5 | The total length of the radio record |

Each of the shade records are fixed length (this is because we need to deal with double byte characters) separated by commas.  There should be the same number of records assigned in the header with one line per shade definition.
| Name |  Length | Comments
| ------ | ------ | ------ |
| Shade Id | 3 | The identifier assigned to the shade 1-32 |
| Paired | 5 | Indicates whether the shade is paired or not.  true or false |
| Shade Type | 3 | Numeric value for the shade type  0 = roller, 1 = blind, 2 = ldrapery, 3 = awning, 4 = shutter, 5 = garage (one button), 6 = garage (3 button), 7 = rdrapery, 8 = cdrapery, 9 = drycontact |
|  Address | 10 | The address assigned to the shade |
| Name | 20 | The name assigned to the shade |
| Tilt Type | 3 | Numeric value for the tilt type 0 = None, 1 = tiltmotor, 2 = integrated, 3 = titlonly, 4 = euromode |
| Protocol | 3 | Numeric value for the protocol 0 = RTS, 1 = RTW, 2 = RTV, 8 = GP_Relay, 8 = GP_Remote |
| Bit Length | 3 | Either 56 or 80 |
| Up Time | 10 | The timing in ms |
| Down Time | 10 | The timing in ms |
| Tilt Time | 10 | The timing in ms |
| Step Size | 5 | The relative size used for step calculations 1-1000 |
| Linked Remote 1 | 10 | The address of the remote |
| Linked Remote 2 | 10 | The address of the remote |
| Linked Remote 3 | 10 | The address of the remote |
| Linked Remote 4 | 10 | The address of the remote |
| Linked Remote 5 | 10 | The address of the remote |
| Linked Remote 6 | 10 | The address of the remote |
| Linked Remote 7 | 10 | The address of the remote |
| Rolling Code | 5 | The last rolling code used by ESPSomfy RTS |
| Flags | 3 | A bitmask of the sensor data |
| My Pos | 12 | The currently stored my position to 5 decimal places.  -1.0 is not set. |
| My Tilt Pos | 12 | The currently stored my tilt position to 5 decimal places -1 is not set |
| Current Pos | 12 | The current lift position of the motor to 5 decimal places 0-100 |
| Current Tilt Pos | 12 | The current tilt position of the motor to 5 decimal places 0-100 |
| Flip Commands | 5 | Indicates whether up is down and down is up.  true or false |
| Flip Position | 5 | Indicates whether 0% is closed or open.  true or false |
| Repeats | 3 | The number of repeats to send when sending a command |
| Sort order | 3 | The sort order to be shown in the interface |
| GPIO Up | 3 | The GPIO pin assigned to the up command if a GP_* type |
| GPIO Down | 3 | The GPIO Pin assigned to the down command if a GP_* type |
| GPIO My | 3 | The GPIO Pin assigned to the my command if a GP_* type |

Next there will be 1 record each for the Groups.  This is a fixed length record separated by commas.

| Name |  Length | Comments
| ------ | ------ | ------ |
| Group Id | 3 | The id assigned to the group  0-16 |
| Group Type | 3 | This is currently only 0 |
| Address | 10 | The remote address assigned to the group |
| Name | 20 | The name assigned to the group |
| Protocol | 3 | Numeric value for the protocol 0 = RTS, 1 = RTW, 2 = RTV, 8 = GP_Relay, 8 = GP_Remote |
| Bit Length | 3 | Either 56 or 80 |
| Member Id 1 | 3 | The id of the shade member |
| Member Id 2 | 3 | The id of the shade member |
| Member Id 3 | 3 | The id of the shade member |
| Member Id 4 | 3 | The id of the shade member |
| Member Id 5 | 3 | The id of the shade member |
| Member Id 6 | 3 | The id of the shade member |
| Member Id 7 | 3 | The id of the shade member |
| Member Id 8 | 3 | The id of the shade member |
[... 48 lines cut ...]
| Board Type | Number | 3 | 0 = Custom, 1 = WT32-ETH01, 2 = Olimex ESP32-POE, 3 = Olimex ESP32-PVB, 4 = LilyGO T-Internet POE, 5 = wESP32 v7+, 6 = wESP32 < v7 |
| PHY Type | Number | 3 |  0 = LAN8720, 1 = TLK110, 2 = RTL8201, 3 = DP83848, 4 = DM9501, 5 = KZ8081 |
| CLK Mode | Number | 3 | 0 = GPIO0 In, 1 = GPIO0 Out, 2 = GPIO16 Out, 3 = GPIO17 Out |
| PHY Address | Number | 4 | Address of the physical adapter 0 or 1 |
| PWR Pin | Number | 4 | GPIO power pin for the adapter |
| MDC Pin | Number | 4 | GPIO pin assigned to the MDC of the adapter |
| MDIO Pin | Number | 4 | GPIO pin assigned to the MDIO of the adapter |

Finally we get to the radio settings.  This is a fixed length record.

| Name |  Length | Comments
| ------ | ------ | ------ |
| Enabled | 5 | Indicates whether the transceiver is enabled or not.  true or false |
| Protocol | 3 | Numeric value for the protocol 0 = RTS, 1 = RTW, 2 = RTV |
[... cut ...]
| MOSIPin | 3 | GPIO Pin for the transceiver |
| MISOPin | 3 | GPIO Pin for the transceiver |
| TXPin | 3 | GPIO Pin for the transceiver |
| RXPin | 3 | GPIO Pin for the transceiver |
| Frequency |  10  | The base frequency assigned to the transceiver to 3 decimal places |
| RX Bandwidth | 9 | The receive bandwidth assigned to the transceiver to 2 decimal places |
| Deviation | 9 | The frequency deviation assigned to the transceiver to 2 decimal places |
| TX Power | 4 | The transmit power for the transceiver |

rstrouse, 2023-11-15:
I am going to close this for now but I also added another wiki to document the backup file.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Backup-File-Anatomy

closed 2023-11-15 by rstrouse