rstrouse/ESPSomfy-RTS #182: MQTT AD device not closing shades

xvisor16, 2023-10-29:
Hi,

A big thanks for adding the MQTT support! After the update the device was immediately added to my Domoticz environment. I did had to use the switch open/close function in Domoticz to get the right button to do the right thing. But one issue remains.
If I press the my/stop button it does what it should goto fav or stop.
If I press the open button when they are down or partially down they go up, when up nothing happens.
So thats all fine but when I press the close button nothing happens. It doesn't matter what I setup in Domoticz close just never works.

So I checked with MQTT Explorer what happens. Domoticz sends a target/set/-100 to close the shades which doesn't work. Manually sending target/set/100 does work. Looking at the MQTT spec home assistent gives it says it should be 100. Now is Domoticz at fault here or has it something to do with the MQTT AD settings ESPSomfy sends?

Thanks for your amazingly quick support and development!! It's really amazing.

Peter

rstrouse, 2023-10-29:
Hmmm... I can't replicate this problem with Home Assistant.  I will review the documentation again to see if there is a setting I missed.

rstrouse, 2023-10-29:
I went through the docs again and it appeared to be correct however, after testing all the shade types again it looks like the docs while technically correct were not clear about the payloads.  Using the Github update choose `Main` for the version and update your device.

xvisor16, 2023-10-30:
I've tried and am running 2.2.1b now, but nothing changed, still only up and my are working, down is not.

It could be that the issue is not with ESPSomfy but with Domoticz, I will create an issue there too and see what they say.

rstrouse, 2023-10-30:
I figured it was a longshot.  I was hoping the inclusion of some of the default values would change the calculus.  If it is sending -100 when the `position_open` is 0 and the `position_closed` is 100 then something is amiss.  The docs indicate that this is the range of values for the target.

rstrouse, 2023-11-15:
@xvisor16 have you figured out why Domoticz is sending a -100 to close the shade?

xvisor16, 2023-11-18:
Hi @rstrouse,
I am currently on Github in contact with gizmocuz from Domoticz he had this under investigation.

jcdel09, 2023-11-22:
hello,
 I am the same problem with domoticz.
Have you solve it?
thank

xvisor16, 2023-11-30:
Hi @rstrouse,

This is the reaction from gizmocuz from Domoticz:

Ahhh well this is incorrent

"position_open": 0,
"position_closed": 100,

it should be

"position_open": 100,
"position_closed": 0,
Which is also the default

https://www.domoticz.com/wiki/Blinds

rstrouse, 2023-12-01:
@xvisor16 please read this section.  https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#invert-position

rstrouse, 2024-01-09:
I am closing this for now as I assume you checked the Invert Position checkbox and it fixed the Domoticz issue.

closed 2024-01-09 by rstrouse