rstrouse/ESPSomfy-RTS #183: Shade does not appear in the dropdown for grouping

lukekaiwalker, 2023-10-29:
Hi,

I have an issue with including shades to a group. The ones which are shown in the dropdown do not fit to the shades which are already included in the group. One shade is missing so that I am not able to include it. Others are shown which are already included.

What can I do?

Thanks, Luke

rstrouse, 2023-10-29:
Please upload a backup here.  I will take a look to see if there is something with your configuration.

lukekaiwalker, 2023-10-29:
Ok, here is the backup file.
[ESPSomfyRTS 2023-10-29T18_52_04.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/13198948/ESPSomfyRTS.2023-10-29T18_52_04.zip)

rstrouse, 2023-10-29:
Which shade does not appear and for which group?

lukekaiwalker, 2023-10-29:
The group is called "EG/Bad ohne Terrasse" and the shade which does not appear is "Bad".

rstrouse, 2023-10-29:
Use the Github update to update your firmware but select `Main` in the version dropdown.  Your issue will go away.  It appears that the list for filtering shades shades that were already members of the group was not being filtered properly.  That dropdown should only show shades that are not already members.

lukekaiwalker, 2023-10-30:
Now it works as intended. I can include the shade to the group... The new update functionality is pretty cool!

Many Thanks!
Luke

closed 2023-10-30 by lukekaiwalker