rstrouse/ESPSomfy-RTS #184: Can’t access ip address interface

Jack123uk, 2023-10-29:
Hey, I seem to be having an issue where everything will work great for so long no more then 2 days, then my blinds will become unavailable, if I try and access the ip address of espsomfyrts it will not connect, the only way to resolve the issue is turning the device off and back on again. I would say it’s like it’s lost WiFi connection but when I look in my router settings it’s still online and available. Do you know what the issue may be? Thanks

rstrouse, 2023-10-29:
Is there any way you can hook it up to a serial connection to see the logs when it fails?  This would mean that you have it plugged into and powered from the serial device.

Jack123uk, 2023-10-30:
Yes the time zone is correct, I’ve reserved the ip address within my router settings is that what you mean? I’ve plugged it into my computer and launched esphome web to see the logs, 
It says successfully connected to ip address, but I can’t access it still, 
And it seems to get stuck on initializing RX Queue

rstrouse, 2023-10-30:
The very next thing it does is call out to github to check for the next release.  This is a great clue.  Can your device reach the internet?  I may need to figure out how to fail the request sooner if that is what is failing.

Jack123uk, 2023-10-30:
Yes I have internet to the device.  After many many reboots I’ve got it all working again, but if it’s following the same trend it will go unavailable again in 24/48 hours. Hopefully not.

rstrouse, 2023-10-30:
What are you powering with?

Jack123uk, 2023-10-30:
An Apple usb charger plug.

rstrouse, 2023-10-30:
It is odd that it is failing like that.  It will be interesting to see the log when it does fail.  I will add some code to timeout if the interweb gets lost.

Jack123uk, 2023-10-30:
Okie thank you.. yes it’s very odd behaviour. I will see how it goes and see if it does it again..

Jack123uk, 2023-10-31:
So i went to update the device today and the same as happened after the update, i cannot access the device from the ip address anymore. connecting it to the logs it stops at loading file /main.css.

Radio Pins Configured!
Successfully set up the radio
Enabled receive on Pin #12 Timing: 1
Initializing RX Queue
[HTTPS] GET... code: 200
[HTTPS] GET... code: 200 - 12884
Socket [0] Connected from (my ip address) url: /
Loading file /index.html
Loading file /main.css

rstrouse, 2023-10-31:
The Loading file /main.css is interesting as the very next thing is should get is a request for the /icons.css file.  I have added a response to the options request just in case the browser asks for it.  The other thing to check is to see if the browser cache is being disrupted.  You will see an error in the developer tools if there is an issue with the browser cache.  There is no error and the there is no follow on request from the browser.

Update the firmware and application from the Main repo and lets see if I see an options request.

rstrouse, 2023-11-02:
@mvantschip what version were you coming from?

rstrouse, 2023-11-02:
@mvantschip the issue that @Jack123uk is having is corruption of the SPIFFS partition.  Although, I think your SPIFFS is corrupt I suspect that it is probably related to a quirk in the update library.  

I have had a few folks that have had the application upload process not write the final bytes to the SPIFFS partition because the code for ending the update reboots the MCU before it has had time to flush.  In v2.2.x I have added a delay after the final bytes to allow it to flush.  In all the hundreds of times I updated the SPIFFS partition it happened only once but I suspect it requires the response to process before it forces the reboot.  This allowed me to understand what was going on because the device was connected to the log at the time.

Unfortunately, you will need to flash the SomfyController.onboard.esp32.bin file then restore the backup that was created when you tried to install the SomfyController.littlefs.bin.  v2.2.x and above should no longer have this intermittent issue.

rstrouse, 2023-11-04:
I am closing this.  The SPIFFS update has been repaired in 2.2.x for boards that do not flush the flash quickly enough before the update library reboots the device.

closed 2023-11-04 by rstrouse

Jack123uk, 2023-11-04:
@rstrouse sorry I’ve been away, thank you for the quick fixes you have made, everything seems to be working perfectly now. Thanks