rstrouse/ESPSomfy-RTS #185: Infinite loading after shade pairing.

Bazzofski, 2023-10-30:
Hi! 
After updating to the latest release (V2.2.0), I've noticed that the interface would start loading indefinitely after hitting "shade paired" on the shade prog menu.
 
[image]

rstrouse, 2023-10-30:
Thanks, I'll have a look.  I suspect that this is a script error and a page reload will make it go away.

rstrouse, 2023-10-31:
If you use the Github update function and install `Main` this issue has been fixed.  It was not sending the linked remotes after pair.

rstrouse, 2023-11-04:
I am closing this for now as v2.2.1 was just released that contains the fixes for this error.

closed 2023-11-04 by rstrouse