rstrouse/ESPSomfy-RTS #186: Instruction how to send logs over mqtt

Hoeby, 2023-10-31:
I am looking for the option to send over the logs, with mqtt.
Is there a possibility to do that?

Why could i like to do that?
I want to use the somfy control to manage other things then a somfy device.
The remote i have is the one with a scroll wheel. Your code is the first one, which works great with this scroll wheel.
I want to use the data that is visible in the logs tab, to send it to node-red and do things with it.

What did i try:
Made a shade and setup it up.
I receive data in mqtt. It show is the shade is moving up or down (on the 1 or -1 data).
But i can't see which button is activated.

rstrouse, 2023-10-31:
If you install the `Main` firmware to your device you will get three additional MQTT topics

```
cmd - The last command sent
cmdSource - internal = command came from ESPSomfy RTS, group = command was issued on the request of a group, remote = an external remote issued the command
cmdAddress - The address of the remote that issued the command.

```

Hoeby, 2023-11-01:
That is a quick update.
Uploade the firmware to 2.2.1b
I tested it, but the "step up" isn't send over mqtt

What did i test and compared it with "logs" in your UI.
command      received in log        received mqtt topic /cmd 
UP                      Yes visible                     Yes visible
Down                 Yes visible                     Yes visible
MY                     Yes visible                     Yes visible
STEP UP             Yes visible                      No Not visible
STEP DOWN       Yes visible                     Yes visible
MY + UP             Yes visible                     Yes visible
MY + DOwn        Yes visible                     Yes visible

Tried inverting the shade in the ESP. Made no difference for "step up".
Still it is showing in the log, but not on MQTT
Also directions and other topics aren't send over MQTT. But those work on version 2.2.0

rstrouse, 2023-11-01:
Step up only exists with 80-bit motors.  That is part of the 3-bytes located in the inter-frame silence.  So if the motor it not defined as 80 bits it has no idea that it needs to decode it.

Hoeby, 2023-11-01:
Could you explaine it more?
I don't get it what it makes the difference in V2.2.0 and V2.2.0b

On V2.2.0 i made a config and saw the direction topic (for example), for the "step up"
Now on V2.2.0b the "step up" direction is gone, without a change on the config.

In both versions i see the "step up" in the log, but this is different with mqtt messages.

This is the log. Where a 80-S bits is written. The module has received a command
[image]

When i compare those signals in nodered, then i only see the "step down".
[image]

rstrouse, 2023-11-01:
Check my previous edit.  The tilt wheel has an interesting movement in that with an 80-bit blind with integrated tilt it first moves the tilt direction and when it reaches 0 on that it starts moving the lift.

Hoeby, 2023-11-01:
Uploaded the new SomfyController.ino.esp32.bin you just put on github
With that version the "step up" and "step down" are working

I only tested the mqtt messages. The ESPsomfy-rts is not going to controll the blinds.
Reason: Unfortunally my somfy blinds died. Have other blinds motors which i want to controll with the same remote.
That is now possible with your great work. Thanks for the effort

rstrouse, 2023-11-01:
If you look there is also a GP_Relay protocol that can trigger up and down with relays to directly control the motor if you need it.  This also works with the step up/down timing and it will trigger with the remote.

Hoeby, 2023-11-01:
Thanks for the information about the GP_Relay protocol.

But i went the long way.
I made a very basic code with an MPU-6050 and EPS32 which measures the angle of the venetian blinds.
When a  motor command is received, the blinds will move to the angle which is requested. This can be a mqtt received command which is trigger by your code

Why this complex solution.
I made my own mechanics for the blinds with a servo motor in it.
But also the manual control still exists, with the ropes. That is why i need the angle sensor, to monitor manual movement.
This manual control is in case of power failure, motor failure. Or just because you are standing close to the blinds and want to change the angle.

Hoeby, 2023-11-01:
Closing this enhancement, because the added changes made it work.

closed 2023-11-01 by Hoeby