rstrouse/ESPSomfy-RTS #189: New integration possibilities.

crackn, 2023-11-05:
Hello, I was looking at some xml files for 2 commonly used somfy interfaces that works over TCP and UDP and tought that you can analise it and see if its possible to implement on ESPSomfy-RTS. it would be great to have support over these integrations as they used to talk to Crestron and Savant systems. 

I am attaching both xml files so you can see if it could be done. 

thanks in advance.

**somfy_mylink.xml**
`<?xml version="1.0" encoding="UTF-8"?>
<component xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:noNamespaceSchemaLocation="racepoint_component_profile.xsd" manufacturer="Somfy" model="myLink" device_class="Shade_controller" minimum_component_engine_version="8.0" rpm_xml_version="3.3" alias="">
    <notes>
        See Somfy myLink profile documentation for more information on integrating Somfy to Savant Systems.
        
        You MUST use Somfy app to setup and configure shade setting. Make sure you are able to control shade the way you want it to on Somfy app because it will work in the same way if you control it through Savant. You will need to rely on the app to get information for setting up Blueprint config.
        
        Notice: After you setup the first myLink for the first time with Somfy app, make sure myLink has a 4 digits pin number. If not, then you will need to redo the setup process again. Delete the MyLink app and download MyLink app for a clean setup. Also, reset myLink by pressing and holding the tiny reset button for about 30 seconds until red light is off.
        
        A SystemID is required to be added in the State Variables drop-down of the unit when inspected in Blueprint. If a "not authorized" error occurs during control, instances have been seen of spaces causing issues. "My ID" for an example, may cause problems, but "MyID" would not. 
        
        To find scene ID:
        1) Go to system monitor.
        2) Select "System State".
        3) Enter "SceneID" in filter.
        4) You will find state name in this format SceneID_&lt;scene name&gt; and it will contain an ID value.
        5) The &lt;scene name&gt; should match to created scene on Somfy app.
        6) Use scene ID on Address 1 on Scene entity.
		====================== Change Log ==================
		Version: 1.3 Change: Updated profile notes for clarity.  Risk: Low Importance: Helpful
		Version: 1.2 Change: Added Scene commands.  Risk: Low Importance: Critical
			Change: Added note for finding Scene ID.  Risk: Low Importance: Helpful
    </notes>
    <component_properties>
        <always_repeat_input_selection>true</always_repeat_input_selection>
    </component_properties>
    <control_interfaces preferred="ip">
        <ip name_on_component="ETHERNET" response_time_length_ms="500" port="44100" protocol="tcp">
            <send_prefix type="character"></send_prefix>
            <send_postfix type="character"></send_postfix>
        </ip>
    </control_interfaces>
    <media_interfaces>
        <data name_on_component="ETHERNET">
            <combined_media>
                <data_media type="ethernet"></data_media>
                <control port="44100"></control>
            </combined_media>
        </data>
        <internal name_on_component="SHADE CONTROLLER">
            <environmental_media></environmental_media>
            <resource resource_type="ENV_SHADECONTROLLER_SOURCE"></resource>
        </internal>
    </media_interfaces>
    <state_variable_list>
        <state_variable name="commandID" min_value="0">0</state_variable>
        <state_variable name="SystemID" user_editable="yes" note="Use the System ID set in Integration tab from myLink app. If have not set the System ID in &quot;Current System ID&quot;"></state_variable>
        <state_variable name="TargetID" owning_logical_component="Shade_controller" state_center_binding="TargetID" state_center_type="string"></state_variable>
        <state_variable name="Power_current_power_Shade Controller" state_center_binding="CurrentPowerStatus" state_center_type="string" owning_logical_component="Shade Controller">OFF</state_variable>
        <state_variable name="Mute_current_mute_Shade Controller" state_center_binding="CurrentMuteStatus" state_center_type="string" owning_logical_component="Shade Controller">OFF</state_variable>
        <volume_state_variable name="Volume_current_volume_Shade Controller" state_center_binding="CurrentVolume" state_center_type="integer" owning_logical_component="Shade Controller" granularity="1" min_value="0" max_value="100">0</volume_state_variable>
    </state_variable_list>
    <logical_component logical_component_name="Shade Controller">
        <implementation>
            <internal name_on_component="SHADE CONTROLLER"></internal>
        </implementation>
        <custom_component_actions>
            <action name="GetSceneList">
                <update_state_variable name="commandID" update_type="increment" update_source="constant">1</update_state_variable>
                <command_interface interface="ip">
                    <command response_required="yes">
                        <command_string type="character">{"jsonrpc":"2.0", "method" : "mylink.scene.list", "params" : { "auth" : "</command_string>
                        <parameter_list>
[... 131 lines cut ...]
                            <parameter parameter_data_type="hex" parameter_data_width="1" parameter_total_width="1">FC</parameter>
                        </parameter_list>
                        <delay ms_delay="1000"></delay>
                    </command>
                </command_interface>
            </action>
            <entity name="Shade" address_components="1">
                <group_representation>
                    <push_button_representation>
                        <release_action name="ShadeUp"></release_action>
                        <osd_press_action name="ShadeUp"></osd_press_action>
                    </push_button_representation>
                    <push_button_representation>
                        <release_action name="ShadeDown"></release_action>
                        <osd_press_action name="ShadeDown"></osd_press_action>
                    </push_button_representation>
                    <push_button_representation>
                        <release_action name="ShadeStop"></release_action>
                        <osd_press_action name="ShadeStop"></osd_press_action>
                    </push_button_representation>
                </group_representation>
            </entity>
        </resource_component_actions>
    </logical_component>
</component>`

rstrouse, 2023-11-05:
Funny, that looks like a uPnP architecture that was tweaked just enough to make it not universal.  I'll look at it and see if there are gems in it.  My guess is that he service descriptions don't match the output though.

crackn, 2023-11-06:
yes, the key here is that other systems use these drivers too so if you support it in espsomfy-rts, it will be amazing.
the idea is to select the option mylink or inteo, and listen and execute the commands acting like the original interface.

we have the protocol: one is TCP and the other is UDP, the ports to listen: 44100 for TCP and 9325 for UDP, 
bunch of parameters, but I think the UP Stop DOWN and targetID (ShadeID) is the basic to begin.
I can test it if needed. 
 
tia

rstrouse, 2024-01-16:
I have a udp interface already built into the software.  I you run the command `http;//upnp.xml` you will see the the device description.  However, I am very limited on program space and the udp outputs would blow that without cutting off a few limbs at this point.

Somewhere there is a coded module though that translates either jsonrpc or the udp packets into commands for the inteo and mylink.  I suspect that these are communicating on a socket.  That is just to high of a hill to climb for me right now.

closed 2024-01-16 by rstrouse