rstrouse/ESPSomfy-RTS #190: After updating to 2.2.1, one shutter is not longer controlled by the group

ElmarJordan, 2023-11-08:
Hello, after updating to 2.2.1, one shutter is not longer controlled by the group. The shutter can be controlled individually as normal, but nothing happens if the group is controlled. The shutter can also not be removed from the group (“unpair group“ is not confirmed by the shutter) in order to reassemble the group. Restoring the backup does not change anything.
Other shutters and groups function without restriction.

rstrouse, 2023-11-08:
Upload your backup file here and tell me which one it is.

ElmarJordan, 2023-11-08:
KZR_Links is not working in the Group Kinderzimmer_Rechts

[ESPSomfyRTS 2023-11-07T10_24_12.backup.zip](https://github.com/rstrouse/ESPSomfy-RTS/files/13298330/ESPSomfyRTS.2023-11-07T10_24_12.backup.zip)

rstrouse, 2023-11-08:
The firmware update should have no bearing on this and your backup file looks fine.  ESPSomfy RTS sends out a single frame and any motors that have the group address in their memor should respond.   It appears the motor simply lost the address from its memory.  So when it asks you if the shade jogged during removal tell it that it did.  Then re-add the motor to the group.

I have seen a couple of instances where users have paired other solutions to a motor but never unpaired them from it.  This fills up the memory and causes the motor to forget things.  Somfy recommends that if you are having pair problems to factory reset that motor and pair it up again.

ElmarJordan, 2023-11-09:
After a factory reset, I was able to restore all functions.
Thank you very much

closed 2023-11-09 by ElmarJordan