rstrouse/ESPSomfy-RTS #191: Add scheduling using an RTC attached

esalgado, 2023-11-09:
Hello!
I am using Home Assistant and came up to this project searching for Somfy programmable blinds. This is awesome!

The blinds are in a friend's house, and he does not have any Home Assistant installed.

I was thinking... How easy would it be to attach an RTC and add a programming function to the web interface so that this can work standalone?

The alternative is install Ha to my friend's computer, but that would be too easy 😂

Thanks!

rstrouse, 2023-11-09:
An RTC is actually not something that I would look to implement as it isn't required because it talks with NTP.  However your request is not without supporters.  I haven't ruled it out but have some other things in the forefront.

https://github.com/rstrouse/ESPSomfy-RTS/issues/80

esalgado, 2023-11-09:
Thanks, since #80 is already open, I close this one!

closed 2023-11-09 by esalgado