rstrouse/ESPSomfy-RTS #192: Movement of blind with integrated tilt incorrect

011V32, 2023-11-15:
When I trigger a movement of the blind with integrated tilt, the movement is carried out correctly, but the integration believes that the movement will continue after the tilt.
Test scenario: Blind is 100% closed and Tilt is 100% closed. Drive blind to 80% and leave tilt at 100%. This raises the blind, then lowers it again to close the tilt, which is done correctly by hardware (although not completely closed, but this is perhaps a different problem). However, the software not only closes the tilt, but also the blind to 100% again.
I made a video about it, which I will send via PM.

011V32, 2023-11-16:
Now it's exactly the other way around. The FW and HW goes up to open, then goes down to adjust the angle, the FW then displays this correctly, but the HW closes completely. (As if the stop command is not executed.)

011V32, 2023-11-27:
Could I help somehow?

rstrouse, 2023-12-09:
Ok so I had some time to troubleshoot and set up two devices so I can see what is sent from the one moving.  It is sending the all the stop commands.  Are you running 2.2.2b from the Main repository?

[image]

rstrouse, 2023-12-16:
Hmm let’s first start by running the GitHub update and selecting main in the version.   This will take you to 2.2.2c and ensure we are both running the same version.

011V32, 2023-12-18:
I have now the version 2.2.2c, but there is still the same behavior.

rstrouse, 2023-12-20:
Upload a configuration backup.  I can't seem to get the same results as you.  The device is emitting all the proper commands to stop the movement.  I have tried multiple configurations and this will be likely something that I missed on the configuration since these days there are many more configuration options.

011V32, 2023-12-22:
I tested it further and the behavior is different depending on where the blind is currently located.
If it is completely closed it works in reality, but not in the simulation. When he's in a hissing position, it's the exact opposite. I made another video about it and sent it to you via PM. Thanks in advance for everything and let me know if I can do anything.

rstrouse, 2023-12-22:
Ok so your backup let me see what was going on.  The settings for the tilt were clearing the direction movement before it sent the my command to stop the lift.  This made the blind move to the my position because it believed it wasn't currently moving.  Pull from main as this issue is the one I am waiting on for v2.2.3,

011V32, 2023-12-22:
I tested it with the actual main version 2.2.2e.
It worked when the tilt position is 53% (my home position), but it don't work when the tilt position is 100%. Then the hardware closes completely and the software think it stops...

rstrouse, 2023-12-23:
Update again from main.  I did not change the version numbers but I am seeing the correct data from the output now.  The check to see whether we were seeking to the end of the movement was not quite right.

011V32, 2023-12-23:
I think I have my Christmas present. 🎁 
Unfortunately I didn't get to test much, but it seems to work almost like it used to.
Only the tilt time is now much shorter than before. But I can adjust that myself in the settings. 👍 
Thanks. Merry Christmas and a Happy New Year.

rstrouse, 2023-12-23:
Yeah strangely enough, this issue was related to the stored my position.  What was happening was that at the edge boundaries for the lift (0% or 100%), it was assuming that the motor was no longer moving before sending the stop command.  So it assumed the stop command was a go to my command.  

I could not replicate this originally because it was about the position being at 100% and as it turns out the check also assumed this at 0%.  It just so happens that in your backup file, the conditions to replicate it were already set up.  It seemed like it was just continuing on and never sent a stop.  But, your stored my position also happened to be at 100%.

There have been some performance improvements related to the processing of the internal data.  This likely accounts for the tilt time response appearing to be shortened.  The improvements trimmed ~140ms off each end of the response so with a 750ms tilt time that represented around a third of the overall timing.

Merry Christmas to you and your family!

rstrouse, 2023-12-28:
I am going to close this for now as I believe we have you squared away.