rstrouse/ESPSomfy-RTS #193: static ip dont seem to apply on ethernet interface (Olimex ESP32-POE)

Strudsen, 2023-11-15:
When I set the "use ethernet" under adapter, the ethernet interface get a dhcp address, and I can access the webinterface through the ethernet. but when I set a static IP, it seem to ignores the static ip at still use a dhcp.

I can see that it save the ip config, and the checkbox "accquire IP Address automatically (DHCP)" is unchecked

Board: Olimex ESP32-POE
FW: 2.2.1

rstrouse, 2023-11-15:
If you still have a reservation for the old address in your router it will use that no matter what based on the mac address.  There really are very few reasons where you might want a fixed IP on a device.  A DHCP reservation for the mac is much more reliable.

rstrouse, 2023-12-31:
I am going to close this for now.  I have tested with another Ethernet board and the IP address assignment worked as advertised.  However, the way this works is a request for the specified IP address and the router can deny it if there is anything in the routing table that says otherwise.

closed 2023-12-31 by rstrouse