rstrouse/ESPSomfy-RTS #194: ESPHome Native API support

nicerloop, 2023-11-19:
Hi,

I recently discovered this project, and I am planing to try and replace my current setup (https://github.com/nicerloop/esphome-somfy-rts) with your solution to get cover position.

I see HomeAssistant integration is handled by a custom integration.

Do you (have implemented | could envision | already rejected) support of ESPHome Native API to expose components to HomeAssistant without custom integration?

rstrouse, 2023-11-19:
I do not.  However, you can also used MQTT discovery and the MQTT integration if you must.  But in the end the native socket integration is so much more efficient and feature rich.

rstrouse, 2023-12-28:
I am going to close this as I don't have any desire to make this into an ESPHome integration.  While it may be a great platform it was pretty obvious early on that the goals of ESPSomfy RTS didn't match with the ESPHome platform.

closed 2023-12-28 by rstrouse