rstrouse/ESPSomfy-RTS #195: Group creation in TUBE curtain (RTV)

sergiosafer, 2023-11-19:
Hi rstrouse, excellent updates that you have been publishing since the last time we spoke. 
I have updated to the latest version 2.2.1 and I can not finish linking the curtains to a group. I have TUBE curtains with RTV protocol and to close the programming of each curtain, I have to hold down the "PROGRAM" button for a few seconds to close the cycle and that works perfectly, but not the same with the groups as it opens the programming from the software but at the time of closing it does not end. I have tried to keep the button pressed for a few seconds but I did not succeed.

Thank you very much

sergiosafer, 2023-11-28:
Hi rstrouse, would you like me to send you some code to get a clearer idea of the problem? I tried several times and the groups are not working for me.

sergiosafer, 2023-12-06:
Hi rstrouse, I hope you are well.
Do you have any advice on how to operate the unit on this curtain model? 
Thank you very much.

rstrouse, 2023-12-10:
Is the issue with the last step?  Does the motor simply not respond when pressing the pair to group button?
[image]

sergiosafer, 2023-12-10:
Exactly, the problem is in the last step. I keep it pressed but it doesn't close the "pair to group" cycle.

rstrouse, 2023-12-10:
Ok it looks like I need to add the hold/repeat code for that to the ui.  I will have a build for you to try shortly.

rstrouse, 2023-12-10:
Update your device with the firmware from the Main repository.  To do this go to the gears section and select the System > Firmware tab.  Then click on the Github Update button and select Main.  Press update and it will install the update.
[image]

You can now hold the prog button and it should allow the Tube motors to link.

sergiosafer, 2023-12-10:
Thank you very much @rstrouse I'll try it and then I'll confirm if it works.

sergiosafer, 2023-12-14:
Hi @rstrouse I did several tests and it doesn't work. I have tried with version 2.2.1 and updated with Main but it still makes the same error. When I try to make the group the Soft opens the program but it never finishes closing the cycle.

rstrouse, 2023-12-14:
Ok lets try this.  I don't have any tube motors so it is possible that these respond to a different command than the prog button.  This is true for some of the garage controllers.   Open the memory for the group by progressing to the Pair to Group step.  Then press the cancel button.  Go to the virtual remote tab and select the motor that you are trying to add to the group.   Start with the down button, then the my button, and then the up button.  If the shade jogs during this time it was added to the group.  You can also try the button combinations as well.

If this works then you can go through the link shade wizard by simply pressing next on the open memory step and answering yes to the prompt asking you if the Pair to Group.  All will be linked and we will know if there is a difference for the Tube grouping.

sergiosafer, 2023-12-15:
Great, I´ll try asap and back to you 
Thanks

rein-10, 2023-12-30:
could you make it work @sergiosafer

sergiosafer, 2023-12-31:
Hi @rstrouse. Unfortunately I have not been able to test as we have had a huge storm that has done a lot of damage and we are working on that. If you want we can close the case and when I get a chance to test it I will get back to you to comment.
Thank you very much as always for your amazing input and happy new year!!!!

rstrouse, 2023-12-31:
@sergiosafer no worries.  I will leave this open for now.  Happy new year and I hope you get back to stasis soon!

sergiosafer, 2024-01-06:
Hi @rstrouse today I have been able to set up the group configuration with the latest version and it works correctly !!!! 
Thank you very much for your patience 
Greetings

rstrouse, 2024-01-06:
Awesome!  Thanks for ***your*** patience.  Happy new year!!!

closed 2024-01-06 by rstrouse