rstrouse/ESPSomfy-RTS #196: Indoor micro lighting radio receiver

open1999, 2023-11-22:
Hello,

I use

https://www.somfy.me/en-gc/products/2401161/indoor-micro-lighting-radio-receiver

It'ok if you put "shadder", but in this case we have 3 buttons.

Dry contact don't run with lighting radio receiver.

Thanks

rstrouse, 2023-11-26:
How does this work.  If you press the up button does it increase the brightness and decrease when you press the down button or is there no brightness involved and up turns it on and down turns it off?

open1999, 2023-11-27:
It's very basic.

"up turns it on and down turns it off"

rstrouse, 2023-12-10:
Update your device with the firmware from the Main repository.  To do this go to the gears section and select the System > Firmware tab.  Then click on the Github Update button and select Main.  Press update and it will install the update.
[image]

There is now a new option for a dry contact 2-button device.
[image]

open1999, 2023-12-11:
Hello

with v2.2.3 on HA we have error.

`Home Assistant Core
2023-12-11 11:37:03.676 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.espsomfy_rts.switch
2023-12-11 11:37:03.695 ERROR (MainThread) [homeassistant.setup] Unable to prepare setup for platform 'espsomfy_rts.switch': Platform not found (Exception importing custom_components.espsomfy_rts.switch).`

I rollback to v2.2.2

open1999, 2023-12-12:
ok

rstrouse, 2023-12-24:
I am going to close this for now as I believe we got you all squared away.

closed 2023-12-24 by rstrouse