rstrouse/ESPSomfy-RTS #197: Support for ESP32-C3.

VagnPetersen, 2023-11-26:
This project is simply amazing!
I got everything running perfectly with an ESP32-WROOM-32 (32-DevKitC). However it is a little bit big and clumsy.
It would be so nice if somebody can provide a firmware file (or guide me on how to compile one for myself) that will support the Xiao Seeed ESP32-C3 board. I know the pin layout is different, but it is easy to remap the GPIO's from the WROOM-board to the Xiao C3-board.

rstrouse, 2023-11-26:
Unfortunately adding a bunch of binary firmware files to the repository would grow it out of proportion,  However, when I post an update it compiles the firmware for various targets.  I re-ran the action from the last update and the resulting files can be found here for the S2, S3, C3 mini variants.

https://github.com/rstrouse/ESPSomfy-RTS/actions/runs/6880964082

VagnPetersen, 2023-11-27:
Thank you for the response.
I tried to flash the .bin file, but it doesn't work :(
I guess the LOLIN C3-mini is a different "animal" and it calls for a boot-loader. Mine has the boot-loader pre-installed and it will flash.

jfabregat, 2023-12-05:
> ESP32-WROOM-32 (32-DevKitC

Thank you foe sharing the model, i only had another model i tested with many ESP32, unitl i saw you comment, went and buy one of those and now it works!!!

rstrouse, 2023-12-23:
I ordered an S3-Mini and a C3-Mini just to see what the differences are.  The C3 only has one core so I will see how it does with the hardware interrupts for the radio.

rstrouse, 2023-12-27:
In v2.2.3 I added onboard binaries for esp32-c3, s2, and s3.  I only have an s3 mini to try so I do not know if the c3 build is perfect.  I have to say that the s3 mini works pretty well with the onboard files.

VagnPetersen, 2023-12-27:
I tried to flash the C3 with the firmware and it seems to work perfect. I haven't connected the radio module to the C3, but the interface works like a charm and I can change all the different parameters.
GOOD work so far. Thank you :)

rstrouse, 2023-12-28:
Thanks for the feedback my C3 is delayed in transit so it will be a while before I get to test it out.  The S3 mini is pretty great so far and is the size of a postage stamp.  It is about 1/5 of the size of the ESP32S.

rstrouse, 2023-12-31:
I have updated the firmware to generate the files for the S3, S2, and C3 when the release is published.  You should now be able to update your C3 to 2.2.4 but you will need to do it manually first using the upload firmware/application buttons.  The binaries are not as compatible as I originally thought between the ESP32 and the ESP32-C3.

The files are in the 2.2.4 release as SomfyController.ino.esp32c3.bin and SomfyController.littlefs.bin for the firmware and application respectively.

closed 2023-12-31 by rstrouse