rstrouse/ESPSomfy-RTS #198: wESP32-PoE Won't Pair

mtwo, 2023-11-29:
Hi! Thanks for providing this project!

I've installed ESPSomfy-RTS on a wESP32-PoE, which I've connected to a E07-M1101D v2.0 board that appears to be identical to the one used in your guide. With this setup, I see logs for my Somfy remote sending commands to my outdoor roller shutter, including the prog command when I press the prog button on the remote. However when I attempt to pair to the shade by pressing PROG in the pairing wizard, the shutter doesn't respond.

I read through some closed issues here and it looks like this typically means that the RX and TX pins are reversed. I _think_ that mine are correct (see photos) and are consistent with the setup that I entered into the radio configuration (screenshotted). I also briefly physically swapped the TX and RX pins to see if that would help and it did not.

Any suggestions? I did see one comment claiming that the pairing process had to be done extremely closely to the shutter controller. I'm standing about one meter away, though there's an exterior wall in between.

[image]
[image]
[image]
[image]
[image]

rstrouse, 2023-11-29:
Just to be sure.  Does your remote have the letters io on it?

mtwo, 2023-11-30:
@rstrouse no, at least I don't think so. It's labeled Somfy Inteo White Single Channel and it fits into a decora wall box / plate.

[image][image]

rstrouse, 2023-12-05:
I don’t have that equipment.  Look at the docs and find a suitable pin for GDO0.  You can get away without hooking up GDO2 and set the RX and TX to the GDO0 pin.

mtwo, 2023-12-05:
No luck: I set RX and TX to the same pins, and tried this on IO4 / pin 15 and IO2 / pin 18. ESPSomfy won't let me set TX to a pin higher than IO30, and I'm assuming that TX requires analog I/O.

With TX and RX set to the same pins, I can still see messages coming from the remote, but ESPSomfy still isn't able to make the motor jog a second time during pairing.

Could this be caused by some kind of weird configuration on the existing remote? For example, if it was already configured with rolling codes or something like that.

rstrouse, 2023-12-06:
Which button do you press on the remote to jog it the second time?

mtwo, 2023-12-06:
I didn't: it would only jog the one time after holding the programming button on the remote for a few seconds. The programming message is recorded in the ESPSomfy radio logs.

However when I'm in the ESPSomfy pairing UI and I hold the remote's programming button until the motor jogs, the instructions tell me to press "prog" in ESPSomfy afterwards and that the motor should jog again to confirm that it's paired. This doesn't happen.

If I tap the programming button on the remote after attempting this, the motor will immediately jog, however I think that this just means that it's ending the pairing process.

rstrouse, 2023-12-06:
You know this can happen if the motor memory is full.  I would reset the motor and try again.  You have all the symptoms.

mtwo, 2023-12-06:
I'll try that!

joaohess, 2023-12-09:
> I'll try that!

I’m having the same problem. Have you found a solution?

mtwo, 2023-12-09:
I’m away for a little while, and I didn’t want to reset the memory on my shutters before I left, just in case they wouldn’t close. I’m going to attempt this in January.

rstrouse, 2023-12-09:
@joaohess are you using a wESP32-PoE?

joaohess, 2023-12-09:
> @joaohess are you using a wESP32-PoE?

Sorry, no, I'm on a ESP-32 WROOM. But I managed to make it work! I'm new to github, so I didn't know closed issues were hidden! I've found the solution on [issue 121](https://github.com/rstrouse/ESPSomfy-RTS/issues/121): I added the URL /shadeCommand?shadeId=1&command=prog&repeat=20 while pairing and it worked!

Thank you

rstrouse, 2023-12-22:
@mtwo did you finally get paired?

mtwo, 2023-12-22:
Still on my trip, I’ll be home in a few days and I’ll attempt this once I have a few days back!

rstrouse, 2023-12-23:
Let me know how it goes.

rstrouse, 2024-02-04:
@mtwo I am going to close this if you are still having pairing issues please open another issue.

closed 2024-02-04 by rstrouse

mtwo, 2024-02-05:
Yep, understood. I've been stymied by travel and bad weather when I haven't been traveling (the shutters are only accessible from outside). Once I have time to investigate further, I'll follow up here if I was able to fix it and I'll open a new issue if I can't. Apologies for the delay!