rstrouse/ESPSomfy-RTS #199: Add gate/double gate as type

MrFl0w, 2023-12-04:
Is it possible to add gate or/and double gate in thé type list with the icon that goes with it?
I have a double gate Somfy RTS, with remote including two buttons. 1 to open both side and 1 to open only the left side.

rstrouse, 2023-12-05:
Yeah I will get to this soon but I have been really busy lately.

rstrouse, 2023-12-09:
I will need you to capture the logs when you press the button to open just one side as well as the button to open both sides.

Let me know which command shows up in the logs for each function.

MrFl0w, 2023-12-09:
The command showing up in the logs for all functions is the same "Toggle".

I have captured the logs and attached them for reference. In that you will get "Open one gate:", "Close one gate:", "Open both gate:" and "Close both gates:".
[RTS Gates.txt](https://github.com/rstrouse/ESPSomfy-RTS/files/13623286/RTS.Gates.txt)

rstrouse, 2023-12-09:
Ok this is like a 1 button garage door except in this case you have motors.  This could be set up as left/right and then grouped for both.  

The relay remote is a single address remote so it cannot address more than one motor at a time.  To move both motors for your gate they are paired to the same channel.  Given the gate operates appropriately with the up and down, I think this is using the same command structure as the Dexxo where you can send either toggle or up/down/my.

If you have the ESPSomfy RTS paired to each motor you can probably see this in action  where up, down, and my all work as they should.  There is another model of the Dexxo motor though that reportedly has to be switched between 1 and 3 button operation at the motor.

I will model the gate control from the garage (3-button) control.

rstrouse, 2023-12-10:
Update your device with the firmware from the Main repository.  To do this go to the gears section and select the System > Firmware tab.  Then click on the Github Update button and select Main.  Press update and it will install the update.
[image]

Once installed there will be an option for Left, Right, and Center gates.

MrFl0w, 2023-12-11:
Great! So I have updated my setup to Main, now I have the version "v2.2.2c".
I have been able to pair the "Gate (center)" without any problem, and it's working fine.
But I am unable to pair the gate left or right... If I am doing so, it will always pair it as a "Gate (center)" to manage both motors.

While looking at the documentation of the gate "Somfy [Exavia 500](https://service.somfy.com/downloads/fr_v5/5073786bexavia_sgs_sav_elec-fr-en-pl.pdf)", to pair the remote control "Somfy [KEYPOP 2 RTS](https://www.somfy.co.uk/products/2401539/keypop-2-rts)" for only one gate to open, the process is the same as for both motors with the difference that we should use the second button of the remote.
And in reality the motherboard of the gate will automaticaly set the single gate based on the second buton of the remote (see page 29 of the notice).

[image]

Also with the new version of ESPSomfy RTS, in homeassistant I have changed to the main version, but it keeps asking me to update it to version 2.2.1...

rstrouse, 2023-12-11:
There was a deployment issue in the HA component.  Please update it to v2.2.3.

According to your logs, the motor has two separate addresses.  The pedestrian gate `5760330` is one address and both gates are another `5694794`.  If you wanted to mirror this function you could pair each of these separately.  The remote could then be linked to the addresses it is paired to for the fob remote.

MrFl0w, 2023-12-11:
Sorry but I am not 100% that I am enough clear, or that I understand you...

Anyhow, I have linked my remote (both button) and the one from my wife. And the addresses are different.

[image]

I really don't know how to deal with that thing....

rstrouse, 2023-12-11:
> Sorry but I am not 100% that I am enough clear, or that I understand you...

The unit has two separate address spaces that it responds to.  One is for both motors and one is just for the pedestrian gate.  Unfortunately, it would have been better if the unit simply allowed you to pair each remote button to one or both of the motors then group them if you wanted both.  However, the functions should work just fine as they are.   If you wanted to pair up just the pedestrian gate you could do that as well with the left or right gate types and link the remote buttons for the pedestrian only gate to that.  Then link the other button to the center gate.

It will not reflect that only one panel is open if the pedestrian gate is opened but it will show the open and closed state of that gate on the other device.  Linking the remote for the both button to the single gate will show the pedestrian gate open when the entire gate opens.

> Anyhow, I have linked my remote (both button) and the one from my wife. And the addresses are different.

Yes every remote will have a unique address as will each device controlled by ESPSomfy RTS.

MrFl0w, 2023-12-12:
I understand now 😉. But the thing is I can't pair the motors one by one....
One Idea would be to unplug the motor cable from the motherboard and try to pair them one by one.
But by following the instructions it doesn't work, as it seems that the buttons a the remote are somehow linked in a way or another.

[image]

rstrouse, 2023-12-22:
I am going to close this for now.  If you have any other questions or need further assistance please open a new issue.

closed 2023-12-22 by rstrouse