rstrouse/ESPSomfy-RTS #201: yellow and red explanation mark for roller shutter image

stas1000, 2023-12-07:
Hi,

Sorry if this is stated in the docs smewhere but i am unable to locate.
Was just wondering if you could shed some light on what the attached screenshot indicates. The shad has change from a normal status to having a yellow and red explanation mark? The last thing i updated before this happened was setting the  MY position on the shade

[image]

Thank You

rstrouse, 2023-12-07:
That is a sensor reading for a windy condition.  You can only do this without a wind sensor from the virtual remote.

stas1000, 2023-12-07:
Is there a way for me to stop it? I checked the virtual remote and both sunny and windy are unchecked

stas1000, 2023-12-07:
Thank you I followed the instructions you gave in the HA forum. Pressed send in the virtual remote with the sensors unchecked.

closed 2023-12-07 by stas1000