rstrouse/ESPSomfy-RTS #202: ESPSomfy-RTS Chronis integration possible?

RainerMaier3, 2023-12-10:
Dear rstrouse, also thanks a lot for your great work.
I have also integrated your ESPSomfy-RTS into my HomeAssistant. It works perfectly :-)
I have one short question. I have in addition to my Remotes (Centralis) also a central remote Somfy Chronis where all Centralis are connected. This was my central control before the Homeassistant integration. Is it somehow possible to connect this Chronis as remote (listener) also to the shades? If the Chronis moved down all shades, it is currently not seen by the ESPSomfy-RTS integration. It would be perfect, if the Chronis actions are also visible in ESPSomfy-RTS as it is for the single remote controls like the Centralis.
Thanks a lot.

MrFl0w, 2023-12-10:
I have also a Telis 1 RTS, for the same purpose. Implementing something that listen to those types of remotes would be fantastic.

rstrouse, 2023-12-10:
You should be able to link the remote today unless I am missing something.   The Telis 1 for sure simply sends the commands to the motor.  ESPSomfy RTS will capture these and reflect the state accordingly.

I would not expect that the schedules on the Chronis ever make it out of the remote.

https://github.com/rstrouse/ESPSomfy-RTS/wiki/Configuring-the-Software#linking-remotes-and-sensors

MrFl0w, 2023-12-10:
Oups I missed that step! Now I have all m'y remotes linked! Thanks

RainerMaier3, 2023-12-10:
Hi,
I have successfully shared the shade and one remote control (Centralis) (see picture)
Now as described in your link, I pressed the button "link Remote" to pair another remote (my Chronis)
unfortunately, nothing happened. The Info Window didn't closed and nothing happened, when I pressed the button "up" of my Chronis, except all shades are moving up.
Did I missed something?
[image]

RainerMaier3, 2023-12-10:
Here is a Photo of my config (other remotes for other shades are in the different rooms) Central unit is located in the Living room.
[image]

RainerMaier3, 2023-12-10:
[image]

rstrouse, 2023-12-10:
When you press buttons on the Centralis remotes do you see any responses in the Radio log for them?

RainerMaier3, 2023-12-11:
Ah, I cannot see a log, even if I am close, I think the signal is out of tune.
I also have seen, that some buttons, that I have pressed are misinterpreted (like an up press is shown as a my in the radio log). Is this also a sign of receiver is a bit out of tune?

RainerMaier3, 2023-12-11:
Hi, that was it,! 
After tuning the radio, I was able to link the Chronis to every shade. Now it works ! 👍 
Thanks you so much 😃 🥳

RainerMaier3, 2023-12-11:
Thanks for this great Work!

closed 2023-12-11 by RainerMaier3