rstrouse/ESPSomfy-RTS #203: Support for Heltec WiFi LoRa 32(V3)

bggsolar, 2023-12-12:
Hi - great project! I'd like to install it to a Heltec Dev Board WiFi Lora ESP32 (V3), do you think there might be a chance to get it running?
https://heltec.org/project/wifi-lora-32-v3/

Still get some issues "buffer overflow" while flashing the the esp onboard version.

rstrouse, 2023-12-12:
I'm sorry.  The only way to do that is to set up a development environment in the Arduino IDE and recompile the binaries for that board.  I don't have one of those and if I had every development board made I would be sitting in a sea of them.  Some vendors do some tweaking to get their branded stuff to work so they may have messed with the bootloader.  On paper it is simply an ESP32-D0 and the specs should allow the firmware to flash.  Buuuut, I am guessing there are tweaks for the power management and the LoRa radio (which you cannot use anyway).

rstrouse, 2023-12-24:
I would suggest you get an ESP32 board without a LoRa radio in it.  You will have success with one of those.   I am going to do some research in the future on the mini boards.

closed 2023-12-24 by rstrouse