rstrouse/ESPSomfy-RTS #204: Cannot connect to ESPSomfy RTS wifi AP

crmtestresco, 2023-12-13:
I am unable to connect to the ESPSomfy RTS wifi.
I tried several computers/phones, but none of them worked:

- Fedora linux - can see the hotspot, but connecting never completes
- Android phone 1 - cannot even see the hotspot
- Android phone 2 - can see the hotspot, but connection to it fails

Hardware I used:

- ESP32-WROOM-32D
- SRAM: 520 KB
- Flash: 4MB
- Dual-core Xtensa® 32-bit LX6 MCU
- 40MHz

I am able to flash the firmware via https://web.esphome.io/ . I can see the logs from the device which say:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
E (240) psram: PSRAM ID read error: 0xffffffff

Startup/Boot....
Connection Type: 1
NTP Settings 
pool.ntp.org TZ:
WIFI Settings
 SSID: [] PassPhrase: []
 Mounting File System...
 File system mounted successfully

 Launching web server...
 Creating Web MicroServices...
 WiFi Mode: 0
 Scanned 6 Networks...
  0: 
  1: 
  2: 
  *3:  (-82dBm) 
  4: 
  5: 

Turning the HotSpot On
Initializing AP for credentials modification

SoftAP IP: 192.168.4.1
```

Wifi networks and MACs have been redacted.
Once I even got to a point where the phone said "Getting IP address"...

Am I doing something wrong? Is there a way to diagnose the issue? Is there some memory requirement on the chip, can it be it has not enough memory?

PS: thank you for this, it looks like a wonderful software!

rstrouse, 2023-12-13:
When your device tries to connect to the AP and just looks like it just didn't finish.  Open a browser and use the http://192.168.4.1 address.

From the specs provided, your device should be fine.  The PSRAM read error is simply the ESP rebooting and shutting down its internal watchdogs.

crmtestresco, 2023-12-15:
The address you mentioned is unavailable. The connection to the network never completes. Connection is unsuccessful and thus reverted to the last working wifi...
Is there a way to diagnose the issue?

rstrouse, 2023-12-16:
Yes hook up the serial connection and see if it gets to the SoftAP IP line once it is there the AP should be active.  Some networks will not let you connect to another one so you may need to disconnect your wifi to see it.

crmtestresco, 2023-12-18:
The logs in my original post is dumped from the serial connection.
It is straight after the firmware install, I did not configure anything yet, it is fresh firmware install.
Sometimes I even see 2 or 3 networks with "*" asterisk in the logs. Which does not make sense, if asterisk means connected network...
I ordered another device, different vendor, will see what it does. Will let you know.

rstrouse, 2023-12-18:
Just to be sure when you initially flash the firmware you need to use on the SomfyController.onboard.esp32.bin not the other files listed in the release.

crmtestresco, 2023-12-19:
That is what I flashed. I even tried different versions to be sure. Will wait for the other HW. Thanks for now

Jacaj66, 2023-12-22:
Hi all.
Same kind of problem for me.
I used latest release v2.2.1 of "SomfyController.onboard.esp32.bin".
Flashed with "Expressif flash_download_tool_3.9.5" following your procedure > ESPSomfy-RTS/wiki/Installing-the-Firmware.

-First difference: I don't find the right hotspot. 
Not "ESPSomfy RTS", but i get "ESP_8B7B6D" AP SSID on one ESP32 module and "ESP_8D3949" on another one.
After connected on those AP's, Windows Wifi parameters gives 192.168.4.1 as IPV4 DNS address and 192.168.4.2 as own PC address.

-With a browser, connection is not possible either on Firefox or Edge or Opera on Windows, with http://192.168.4.1 giving "ERR_CONNECTION_REFUSED" error after timeout.
Same problem with my android phone.

-So, no gear icon at the upper right to open the configuration screen...
Not sure but DNS address and EspSomfy server on the same address, is that normal ?
Tested on 2 different modules from: "https://www.az-delivery.de/en/products/esp32-d1-mini"  
> Esp32 Wroom with Chipset: ESP32-D0WDQ6.

Any clues ? Thank you.

rstrouse, 2023-12-22:
Unfortunately, not all the flashers are equal and neither are the dev boards.  You may be able to flash with one of the firmwares from here 
https://github.com/rstrouse/ESPSomfy-RTS/actions/runs/7303655773
but the sheer number of available boards that have been tweaked by others makes it impossible to build firmware for all versions.   In most cases they just work but a lot of them tweak their bootloader to add features.  While the D0WDQ6 should flash fine with the current firmware the bootloader can be changed by the carrier board manufacturer.

Jacaj66, 2023-12-23:
Hi rstrouse. 
I solved 2 problems (for the moment) 

-1) Bad AP SSID and connection problem to access the first configuration screen:
It was a beginner problem. (what I really am) 
In Expressif flash_download_tool_3.9.5, i forgot to check the little box to the left of the file name (.bin). 
In fact, the flashing was not really done, the flash program did something when i used the "Start" button, but I don't know what.
Then the bad SSID's detected were undoubtedly contained in the base firmware of ESP32.
(About the alternative .bin files for different ESP32 you suggested, I think you also had to give the loading addresses for the four .bin files to be usable by the flash program, but I haven't had to use them.)

-2) After accessing the wifi's SSID definition screen, entering its password and disconnecting from 192.168.4.1, I still couldn't connect to the new DHCP address assigned to the ESP32, with the same time out.
In fact, this was due to multiples connections of the ESP32 to my PC.
The dual Wifi+serial connection via USB at the same time was the source of the Wifi/browser access problem.
Once the ESP32 had been powered by a simple USB charger, the Wifi connection and the rest of the configuration could be carried out. 

Now, I'm waiting for the radio module to be delivered before continuing with the setup.
I hope I've made myself clear, as English is not my native language.

Thank you very much for your help.
Jacques.

rstrouse, 2023-12-23:
The espressif flash download tool does require knowing the bootloader address as well as the initial PSP address.  I looked high and low for a definitive list of values for such things.  But suspect they are buried somewhere in the boards file.  I have long suspected that the address starts at either 0x1000 or another unknown address that I don't know about.  Definitive answers on that question currently eludes me. 

I ordered some boards to try out as I have never flashed from the artifacts but other more experienced with the target boards have.  Perhaps I can get a better understanding of how to assemble these files into a single onboard file starting at address 0x000.

I am guessing the bootloader is at 0x1000, partitions at 0x8000, littlefs at 0x290000 and some combination of the application at 0x10000.  Perhaps some more seasoned esp32 hw folks may come along and clear this up.

crmtestresco, 2023-12-23:
Hi, I obtained different device. Everything worked fine on the first try. Flashed, rebooted, connected to AP, browsed the GUI.
I have no idea what is wrong with the first device, nor I know how to debug that. So we can close this issue if you ask me.

Jacaj66, 2023-12-23:
About the .bin files start address, i found a kind of explanation in the following video at 1mn30   
https://www.youtube.com/watch?v=EGgqdD4oU0U.  [ESP32 Flash Download Tool Usage on Windows]
Addresses are supposed to be given in a readme.pdf when compiling with Eclipse, as explained if you activate subtitles in the video. It's not very clear to me, but certainly much clearer to you.

rstrouse, 2023-12-24:
@Jacaj66 thanks for the link.  I'll check it out.

closed 2023-12-24 by rstrouse

HansVanEylen, 2024-01-05:
I had similar issues as the op : none of the Android devices in the house could stay connected to the AP long enough to load the page at http://192.168.4.1. They all switched back to their 'normal' network, sometimes immediately, sometimes after 30 seconds.
I was finally able to get to the page by using an older iPad. Just wanted to add this here as it might help other people.