rstrouse/ESPSomfy-RTS #205: Homeassistant Error espsomfy_rts.switch

xxLexanixx, 2023-12-13:
Logger: homeassistant.setup
Source: setup.py:403
First occurred: 15:32:18 (1 occurrences)
Last logged: 15:32:18

Unable to prepare setup for platform 'espsomfy_rts.switch': Platform not found (Exception importing custom_components.espsomfy_rts.switch).

rstrouse, 2023-12-13:
Reload the integration by clicking on the reload menu in HACS.  This should no longer be happening and was caused by a whiffed deployment on my part.

gieljnssns, 2024-01-01:
@rstrouse 
I'm still seeing an error likte this every HA restart
```
Logger: homeassistant.loader
Source: loader.py:842
First occurred: December 31, 2023 at 12:45:11 (1 occurrences)
Last logged: December 31, 2023 at 12:45:11

Unexpected exception importing platform custom_components.espsomfy_rts.switch
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 842, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 859, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 936, in exec_module
  File "<frozen importlib._bootstrap_external>", line 1074, in get_code
  File "<frozen importlib._bootstrap_external>", line 1004, in source_to_code
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/espsomfy_rts/switch.py", line 28
    if "shadeType" in shade and (int(shade["shadeType"]) == 9 or int(shade["shadeType"] == 10):
                                                                                              ^
SyntaxError: invalid syntax
```
```
Logger: homeassistant.setup
Source: setup.py:403
First occurred: December 31, 2023 at 12:45:11 (2 occurrences)
Last logged: December 31, 2023 at 12:45:12

Unable to prepare setup for platform 'espsomfy_rts.switch': Platform not found (Exception importing custom_components.espsomfy_rts.switch).
```