rstrouse/ESPSomfy-RTS #206: IO-Remote LOW/HIGH

sn3me, 2023-12-17:
Would it be possible to change the low/high status of the GPIO's for controlling a somfy io remote?

set the 3 GPIO's to high as default
trigger pushing a button by setting to respective GPIO to low for 0,1s

rstrouse, 2023-12-17:
Hmm you know that during boot you will get a period when all the pins are low or float.  If you are powering the relay with the same juice it will be triggered during boot unless you pull them high with an external pullup.  

The trigger length is controlled by the number of repeats to about 170ms (normal frame length) per repeat.

I can certainly change the triggerGPIOs method to invert high and low to accommodate but it does come with warts in that many of the pins on esp32 are pulled low at boot or they float.  Which relay are you using for reference or are you triggering buttons on a dissected remote?

sn3me, 2023-12-19:
It's a dissected remote. Comparable to this one https://projecthub.arduino.cc/beejayf/somfyduino-io-2def54
the mentioned trigger length of 170ms should not be an issue

sn3me, 2023-12-21:
that works. 
great - many thanks

rstrouse, 2023-12-22:
I am going to close this one for now.  This functionality is included in the next release but you can continue to run from the main repository.

closed 2023-12-22 by rstrouse