rstrouse/ESPSomfy-RTS #208: [feature request] Publish radio log entries as MQTT message

paede81, 2023-12-22:
Hi

As far as I see, it's currently not possible to get logs events via MQTT. I think it would make sense to provide a option to enable the forward of radio log entries as MQTT message. That would it make easier to monitor the Somfy devices via a 3rd party application like Home Assistant or NodeRed

rstrouse, 2023-12-22:
MQTT does in fact emit the last command processed and where it initiated from repeats are not sent because the rapid succession of this can choke subscribers.  MQTT is not terribly efficient and when you have 48 devices dispatching on a pub channel it saturates quickly.  I have had to really limit the pub functions, otherwise it tends to overwhelm subscribers like FHEM.    

However, for something like NodeRed you will get the dispatches for each processed frame and for HA that information is in the attributes of the entity.

I see a number of HA issues in the HA blog related to performance that I suspect are devices publishing MQTT to the broker and the subscriber code not being able to keep up with the pump from the broker socket.

ESPSomfy RTS does not store logs on the device and will not do that since the storage requirements would be wasted for other much needed items.

rstrouse, 2023-12-30:
I am closing this for now.  I suspect you have found the MQTT messages that you were looking for,

closed 2023-12-30 by rstrouse