rstrouse/ESPSomfy-RTS #209: ESP32-S2: Service Error - Out of memory parsing JSON - when try to add new Somfy device

paede81, 2023-12-22:
Hi
I'm currently running ESPSomfy-RTS on a Wemos S2 mini (ESP32-S2). When I try to add a new Somfy device with a name longer then 8 character, I get the following message:

[image]

Name: `StorenOb` --> working
Name: `StorenObe` --> error

And sometimes the device also crash afterwards and needs a reboot

rstrouse, 2023-12-22:
I will have a fix shortly.  That's what I get for not allocating more memory than the shade definition needs.

rstrouse, 2023-12-22:
This has been fixed in the Main repository.  I allocated more bytes for the json deserialization.

paede81, 2023-12-23:
Thanks a lot. Issues fixed for me (thought it was due to the ESP32-S2 which has less memory than the ESP32)

closed 2023-12-23 by paede81